yaacov/node-modbus-serial #187: Set a bit of an holding register

robertsLando, 2018-02-12:
I have been unable to find a modbus TCP server/slave with FC 22 and 23 implemented so far. The goal of FC22 is to be able to write to holding registers single bit as in the PLC world for example %MW1.1

> The MODBUS service FC22 “Mask write” is a register service that makes it possible to specifically change bits in a register.

Any hint if it already exists or I will have to implement this myself ? 

The alternative is to implement it by creating a function like 'setRegisterBit(reg, bitIndex, bitValue, callback)` that make 2 operations:

- Read register `reg`
- Convert the value read :

```
value = value.toString(2);
value = value.length < 16 ? '0'.repeat(16-value.length) + value : value;
value = value.split('').reverse();
value[bitIndex] = bitValue;
value = value.reverse().join('');
value = parseInt(value, 2);
```

What about this feature?

yaacov, 2018-02-12:
we currently support:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L310
https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp.js#L109

I am not against supporting extra FC's, you are welcome to add a pull request that adds this functionality.

robertsLando, 2018-02-12:
FC 22 is really poorly supported so I think I will go for the second alternative.

closed 2018-02-12 by robertsLando