yaacov/node-modbus-serial #188: CRC error on readInputRegisters

Athiththan, 2018-02-27:
I have been started working on reading multiple input registers from energy meters. I am getting an error on CRC check Failed when connecting and reading Input Registers ... (Still a beginner)

What will be the issue 

please help me, thanks in advance

yaacov, 2018-02-27:
Hi, thanks for the issue,
Can you post the code that fails ?

Athiththan, 2018-02-27:
setInterval(() => {
    master.readInputRegisters(config.Address, config.Length, (err, data) => {
        if (err)
            console.error(colors.red.bold("Modbus Read Input Registers Error | Exception: " + err));
        else {
            console.log(data);
        }
    });
}, timeOut.timeout);

this is the part of the portion i am working on it

yaacov, 2018-02-27:
where did `master` come from ?

Athiththan, 2018-02-27:
const modbusSerial = require('modbus-serial');
const master = new modbusSerial();
master.connectRTU(config.Port, {
    baudRate: config.BaudRate,
    dataBits: config.DataBits,
    stopBits: config.StopBits,
    parity: config.Parity
});

//stopBits: config.StopBits
//rtscts: true

master.setID(config.SlaveID);

yaacov, 2018-02-27:
try `connectRTUBuffered` instead of `connectRTU`

Athiththan, 2018-02-27:
if i use `connectRTUBuffered` it outputs none

yaacov, 2018-02-27:
do you have good readings using other modbus client ?

Athiththan, 2018-02-27:
Yes the same code worked for 2 other energy meters from another vendor. This meter is different from the others 2 so i tried all the possible ways to connect to it. But it throws me the Exception of CRC failed

yaacov, 2018-02-27:
sounds like the meter is faulty

Athiththan, 2018-02-27:
or is it regarding any bit arrangement

Athiththan, 2018-02-27:
Thank you ... for the quick responses bro

closed 2018-02-27 by Athiththan