yaacov/node-modbus-serial #199: Node environment error for version 7.10.0

vasukumar169, 2018-04-26:
Hi,

I am unable to setup the environment on the version 7.10.0.

It is working fine on 8.9.4 version.

I am also attaching the screen shot for your reference in windows.

[image]

I am also attaching the screen shot for your reference in Linux.

[image]

Could you tell , will this package supports 7.10.0 version or  it only supports 7.10.0 version.

yaacov, 2018-04-26:
Thanks !
I hope to look at it this weekend, if you find out more about this, please comment here.

vasukumar169, 2018-04-27:
I just install v6.10.3 as well, it also shows same error as it was earlier, I am also attaching my piece of code.

[image]

The below code is working fine only in v8.9.4

/******************CODE BELOW *********************************/

var ModbusRTU = require('modbus-serial');
var client = new ModbusRTU();
client.connectRTUBuffered("COM4", { baudRate: 19200 }, write)

function read() {
	client.readHoldingRegisters(47,1)
	  .then((resp)=>{
			console.log("resps")
			console.log(resp)
		})
	  .catch((error)=>{
			console.log(error)
		});
	}

function write() {
	client.setID(1);
	manageData()
}

function manageData () {
  client.writeRegisters(46, [0xA108])
	 .then((resp)=>{
		 console.log("red")
		 setInterval(read, 1000);
	 	l})
		.catch((error)=>{
			console.log(error)
		})
}

yaacov, 2018-04-27:
From the logs it looks like your problem is the binaries of serialport you are using.

It looks like serialport should work with your setup:
https://www.npmjs.com/package/serialport#platform-support

Try to remove all node-modules and then re-install using different node versions, it looks like you installed using one node version, and try to run using different one.

closed 2018-05-02 by vasukumar169