yaacov/node-modbus-serial #200: synchronous style

virgo888, 2018-05-02:
how to use modbus-serial with synchronous style ?

virgo888, 2018-05-03:
Hi, thank you for your answer.

`const ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
client.connectRTUBuffered("COM3", { baudRate: 9600 });
var async = require("async");

setInterval(function(callback) {
    async.parallel({
            async_modbus_1_1: function(clbk) {
                client.writeFC3(1, 4103, 2, function(err, data) {
                    console.log(data);
                });
            },
            async_modbus_1_2: function(clbk) {
                client.writeFC3(1, 4106, 2, function(err, data) {
                    console.log(data);
                });
            },
            async_modbus_2_1: function(clbk) {
                client.writeFC3(3, 4103, 2, function(err, data) {
                    console.log(data);
                });
            },
            async_modbus_2_2: function(clbk) {
                client.writeFC3(3, 4106, 2, function(err, data) {
                    console.log(data);
                });
            }
        },
        function(error, result) {
            if (error) {
                console.log(error);
            }

            if (result) {
                console.log(result);
            }
        });
}, 1000);`

I try with this code but cant reply the value, can you help me? sorry cz I'm a newbie in node js.

thanks.

yaacov, 2018-05-03:
Hi,

Serial modbus is serial ( requests are done one **after** another ) .

**You can not do parallel requests on a serial connection,** not even on a serial over TCP connection.

You have to wait for an answer ( or timeout ) of first request before you can send the next one.
[ It is not advised (*), but you can do this on a TCP connection but never on serial one. ]

(*) the hardware may be implemented in a way that does not like multiple asynchronous requests.

yaacov, 2018-05-03:
example of calls waiting for first one to finish, in this example `readCoils` is called only after `readRegisters`  is finished.

``` js

function readRegisters() {
    // read the 4 registers starting at address 5
    client.readHoldingRegisters(5, 4)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .catch(function(e) {
            console.log(e.message); })
        .then(readCoils);
}

function readCoils() {
    // read the 4 registers starting at address 5
    client.readCoils(1, 20)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .catch(function(e) {
            console.log(e.message); })
        .then(close);
}
```

virgo888, 2018-05-03:
Thanks yaacov for your answer, 

but I have an example like this : 

```
const SerialPort = require('serialport');
const { ModbusMaster, DATA_TYPES } = require('modbus-rtu');

var async = require('async');

const serialPort = new SerialPort("COM4", {
    baudRate: 9600
});

const master = new ModbusMaster(serialPort);

setInterval(function() {
    var modbus1 = [];
    var modbus2 = [];

    async.parallel({
            async_modbus_1_1: function(callback) {
                // 4103 Float 1 [oil]
                master.readHoldingRegisters(1, 4103, 2, DATA_TYPES.HEX).then((data) => {
                    // console.log(data);
                    var get_data = data[0];

                    var get_data_a = '0x' + get_data[0] + get_data[1];
                    var get_data_b = '0x' + get_data[2] + get_data[3];
                    var get_data_c = '0x' + get_data[4] + get_data[5];
                    var get_data_d = '0x' + get_data[6] + get_data[7];

                    var get_data_CDAB = Buffer.from([get_data_c, get_data_d, get_data_a, get_data_b]);
                    var get_data_CDAB_float = get_data_CDAB.readFloatBE().toFixed(2) // set 2 decimal
                        // console.log(get_data_CDAB)
                    console.log('MODBUS ID 1 => OIL : ' + get_data_CDAB_float);
                    modbus1.push(get_data_CDAB_float);
                    callback(null, get_data_CDAB_float);
                }, (err) => {
                    console.log(err);
                });
            },
            async_modbus_1_2: function(callback) {
                // 4106 Float 2 [water]
                master.readHoldingRegisters(1, 4106, 2, DATA_TYPES.HEX).then((data) => {
                    // console.log(data);
                    var get_data = data[0];

                    var get_data_a = '0x' + get_data[0] + get_data[1];
                    var get_data_b = '0x' + get_data[2] + get_data[3];
                    var get_data_c = '0x' + get_data[4] + get_data[5];
                    var get_data_d = '0x' + get_data[6] + get_data[7];

                    var get_data_CDAB = Buffer.from([get_data_c, get_data_d, get_data_a, get_data_b]);
                    var get_data_CDAB_float = get_data_CDAB.readFloatBE().toFixed(2) // set 2 decimal
                        // console.log(get_data_CDAB)
                    console.log('MODBUS ID 1 => WATER : ' + get_data_CDAB_float);
                    modbus1.push(get_data_CDAB_float);
                    callback(null, get_data_CDAB_float);
                }, (err) => {
                    console.log(err);
                });
            },
            async_modbus_2_1: function(callback) {
                // 4103 Float 1 [oil]
                master.readHoldingRegisters(2, 4103, 2, DATA_TYPES.HEX).then((data) => {
                    // console.log(data);
                    var get_data = data[0];

                    var get_data_a = '0x' + get_data[0] + get_data[1];
                    var get_data_b = '0x' + get_data[2] + get_data[3];
                    var get_data_c = '0x' + get_data[4] + get_data[5];
                    var get_data_d = '0x' + get_data[6] + get_data[7];

                    var get_data_CDAB = Buffer.from([get_data_c, get_data_d, get_data_a, get_data_b]);
                    var get_data_CDAB_float = get_data_CDAB.readFloatBE().toFixed(2) // set 2 decimal
                        // console.log(get_data_CDAB)
                    console.log('MODBUS ID 2 => OIL : ' + get_data_CDAB_float);
                    modbus2.push(get_data_CDAB_float);
                    callback(null, get_data_CDAB_float);
                }, (err) => {
                    console.log(err);
                });
            },
            async_modbus_2_2: function(callback) {
                // 4106 Float 2 [water]
                master.readHoldingRegisters(2, 4106, 2, DATA_TYPES.HEX).then((data) => {
                    // console.log(data);
                    var get_data = data[0];

                    var get_data_a = '0x' + get_data[0] + get_data[1];
                    var get_data_b = '0x' + get_data[2] + get_data[3];
                    var get_data_c = '0x' + get_data[4] + get_data[5];
                    var get_data_d = '0x' + get_data[6] + get_data[7];

                    var get_data_CDAB = Buffer.from([get_data_c, get_data_d, get_data_a, get_data_b]);
                    var get_data_CDAB_float = get_data_CDAB.readFloatBE().toFixed(2) // set 2 decimal
                        // console.log(get_data_CDAB)
                    console.log('MODBUS ID 2 => WATER : ' + get_data_CDAB_float);
                    modbus2.push(get_data_CDAB_float);
                    callback(null, get_data_CDAB_float);
                }, (err) => {
                    console.log(err);
                });
            }
        },
        function(error, result) {
            if (error) {
                console.log(error);
            }

            if (result) {
                // console.log(result);
                var resModbus = result;

                var modbus = [modbus1, modbus2];
                console.log(modbus);
            }
        })

}, 1000);
```

the code runs fine with the async parallel, but I dont understand, can you tell about that, I'm still learning about the modbus and the nodejs. sorry if I ask a lot. thanks before.

yaacov, 2018-05-03:
modbus-rtu is a different pacage :-)
the example above is not using modbus-serial :-)

you should ask the modbus-rtu people.

virgo888, 2018-05-04:
So modbus-serial package can't create like that structure code ? using async?

yaacov, 2018-05-04:
> So modbus-serial package can't create like that structure code ? using async?

No, you can look at the examples to see how to use modbus-serial:
https://github.com/yaacov/node-modbus-serial/tree/master/examples

The code you pasted above does not use modbus-serial.

virgo888, 2018-05-07:
Ok, thanks yacoov, for your help.

closed 2018-05-07 by virgo888