yaacov/node-modbus-serial #207: Feature request: FC 20, 21

tmsound, 2018-05-29:
I don't know how many others would want this but I have been running into some hardware that is using FC 20 (read file record) and FC 21 (write file record) to change some pretty important settings.  Aside from that, the coverage you guys already have is awesome.  I pre-apologize if this was an inappropriate place to feature request.

yaacov, 2018-05-30:
Hi, thanks for the issue.

I do not think I will have time for that in the next few weeks, If someone wants to take this It will be nice.

@tmsound 
If you take this, please comment here.

yaacov, 2018-06-24:
@tmsound hi, I'm closing this for now, if you do this, reopen this and link your pull request here ...

closed 2018-06-24 by yaacov