yaacov/node-modbus-serial #210: Server is not processing parallel requests

dushyantbangal, 2018-06-04:
I just started the example server

I created a client that sends more three `readHoldingRegisters` requests.

However, on the server, some requests are not being received. This is not an issue with the client as I have tested it with a simulator and it works fine.

I've just put multiple requests in the example client logger.

```
function run() {

    // read the 4 registers starting at address 5
    client.readHoldingRegisters(5, 4)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .then(function() {
        	setTimeout(run, 1000); })
        .catch(function(e) {
            checkError(e);
            console.log(e.message); });
            
     client.readHoldingRegisters(10, 4)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .then(function() {
        	setTimeout(run, 1000); })
        .catch(function(e) {
            checkError(e);
            console.log(e.message); });
     client.readHoldingRegisters(15, 4)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .then(function() {
        	setTimeout(run, 1000); })
        .catch(function(e) {
            checkError(e);
            console.log(e.message); });
}
```

On the server, the `getHoldingRegister` does not get executed for all of the addresses.

yaacov, 2018-06-04:
@dushyantbangal Thanks for the issue :exclamation: 

I will not have time to look into it in the next few weeks, can you try and fix it ?

dushyantbangal, 2018-06-06:
@debevv can you give some brief info why it's happening?

debevv, 2018-06-06:
The problem lies in servertcp.js:808, inside the socket "data" event handler. The code is assuming that the received data will be always of length 12 (MBAP + PDU), but with OP's client code or this one
```
client.writeCoil(5, true)
client.readCoils(5, 1)
client.readDiscreteInputs(5, 1)
```
the data sent with requests 2 and 3 is received packed inside a buffer of length 24, which is normal if the time between the two request is short enough.
This line copies the entire chunk (skipping MBAP) inside the requestBuffer, without stopping after 8 bytes:
```
data.copy(requestBuffer, 0, 6);
```
which will make check in `_errorRequestBufferLength()` fail. So you are basically losing 2 requests here.
One solution would be forcing `data.copy()` to take just 8 bytes from the chunk, and then putting the entire handler in a loop, in order to be able to extract multiple requests at once.
Anyway it is better to never assume the size of `data` in these cases, so I would also add a buffer which holds "remaining" data that cannot be parsed because not long enough (maybe split across 2 TCP packets), which can be parsed at the next "data" event.

debevv, 2018-06-06:
Wait, I forgot that PDU won't be always of 8 bytes, we should read the message length field inside the MBAP and act accordingly.
So, why is the check inside '_errorRequestBufferLength()' forcing a length of 8 bytes?

dushyantbangal, 2018-06-08:
Hasn't @yaacov done a fix for similar kind of issue on the client side in the buffered ports?

yaacov, 2018-06-08:
> Hasn't @yaacov done a fix for similar kind of issue on the client side in the buffered ports?

@dushyantbangal I do not remember :-) but it's ok to just fix it here ... as needed.

pkaz, 2018-06-22:
Hello everyone, 
From what I have noticed, the client has trouble executing multiple requests the way it is being shown further above. 

For me this is the code that worked on the client side: 

`let results=[];
results.push(await client.readInputRegisters(0, 12));
results.push(await client.readInputRegisters(20, 2));
results.push(await client.readInputRegisters(28, 12));
results.push(await client.readInputRegisters(46, 2));

results.forEach(result => {
// DO STUFF HERE WITH RESULTS
});
`

I am trying to read from a modbus device, not a server.

yaacov, 2018-06-22:
@pkaz nice, liked the await syntax :-)

a. can you make a small runnable example with await (it's cool)
b. your device is tcp/rtu/... ?
c. does it always work with 1 request and start failing on the 2nd request , or is it random, sometimes fails on 3rd, 5th etc ... ?

debevv, 2018-06-22:
@pkaz can you try the same example after applying my PR? https://github.com/yaacov/node-modbus-serial/pull/216

yaacov, 2018-06-24:
@dushyantbangal @debevv hi, is this issue closed by #216 ?

debevv, 2018-06-24:
According to my tests it can be closed, but it would be even better if @pkaz and @dushyantbangal could test it against their applications/snippets as final confirmation

yaacov, 2018-06-24:
> but it would be even better if @pkaz and @dushyantbangal could test it against their applications/snippets as final confirmation.

:+1: 

p.s. I think @pkaz problem is not connected to the server side, and he didn't say if he use tcp or rtu to connect to his controller ... , @pkaz ?

dushyantbangal, 2018-06-25:
Hi, Sorry just saw the thread. I'll test it today and see if its working in my case

dushyantbangal, 2018-06-25:
@yaacov @debevv 
@pkaz is saying 
> For me this is the code that worked on the client side:
So I think he was proposing a solution; he didnt face the issue with his script.

However, his solution includes `await` so technically those are serial requests, and not parallel.

dushyantbangal, 2018-06-25:
I've tested the issue with my script, and it looks like its been fixed!

closed 2018-06-25 by dushyantbangal

yaacov, 2018-06-25:
@dushyantbangal Thanks :+1: 

@pkaz I hope to have time next weekend to look at your issue ( it looks different than this one ) if you think I forgot ... just open an issue specific to the problem you are seeing.

pkaz, 2018-06-26:
So I will try to answer everything, ( I am really sorry for the long weekend )...
@yaacov 

- a. a sample client would go like the code below. 
- b. It connects to a Serial over TCP/IP (encapsulated) device.
- c. If I try to read Input Registers on multiple requests, it always fails on the second and never answers.

### Sample client

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
client.connectTelnet("10.10.10.159", {port:4001});
client.setID(1);

setInterval(async function() {

  let results=[];
  results.push(await client.readInputRegisters(0, 12));
  results.push(await client.readInputRegisters(20, 2));
  results.push(await client.readInputRegisters(28, 12));
  results.push(await client.readInputRegisters(46, 2));
  // console.log(results);

  results.forEach(result => {
    for (let i=0; i<result.buffer.length; i+=4){
      console.log(result.buffer.readFloatLE(i));
    }
  });

}, 500);

```

@debevv 
I will pull the new version that got merged and try again later today. But as I said, mine was a client side problem... 

@Everyone
I have found out that if I try to Read more than 21 Registers there is an exception as well, that is why I went with the multiple request solution with await. @dushyantbangal  is right, they are not parallel request... I would like to get the full length of registers in one request.

dushyantbangal, 2018-06-27:
@pkaz, It might be possible that whichever device you are reading from does not allow more than 20 registers in one request. I have come across some meters that limited reading 40 registers in a single request. So I dont think that is fixable from this side, its the manufacturer.

Regarding the parallel issue, it is possible that your TCP-RTU gateway is messing up as thinking its all a single request (happened to me). Try putting a small delay between 2 reads - like 50 ms. They still will be kind-of parallel requests as you are not waiting for one to complete before firing the next one.

Try `client.connectTcpRTUBuffered` instead of `client.connectTelnet` if possible.

pkaz, 2018-06-27:
@dushyantbangal Tried it (connectTcpRTUBuffered), the promise never responds... :(  I think I am doing something wrong...

dushyantbangal, 2018-06-28:
@pkaz continue this discussion at https://github.com/yaacov/node-modbus-serial/issues/226