yaacov/node-modbus-serial #217: setID/Address reading

avnerkhan, 2018-06-22:
Hello , 

I have a computer connecting to a TCP/IP connection  and  this TCP/IP connection is connected to by RTU (thru GridConnect) to 1 Modbus slave device.  I want to eventually have multiple slave devices connected to the master, and to be able to read a register from each slave. I was planning to do this by having the slaves have different Addresses (1-247) , and the client.setID() function would allow me to pick which device I read from by the address number. Currently, my slave device is set to address 1, and it is reading registers fine. However, if I set the client ID to anything other than 1, it still reads fine. I was expecting this to not working, as it should be looking for a another device ID that currently does not exist. Furthermore, if I change the Modbus device's address to anything other than 1 , then the program cannot read registers anymore no matter what I change the client ID to in my program. I have tried changing the transactionIdWrite, but that didn't do anything. 

I am currently using client.connectTcpRTUBuffered() . Any help would be appreciated.

yaacov, 2018-06-22:
Hi, Thanks for the issue,

a. do you have a device that connect directly to tcp, ( to check if the problem is with GridConnect ) ?
b. did you get same problems with `connectTCP` ?
c. can you run with debug on `export DEBUG=modbus-serial` see example [1] and look at the data frame sent [2] , looking for the TCP frame unitID ?
d. do you have some other modbus client to check with ?

[1] https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js
[2] https://en.wikipedia.org/wiki/Modbus#Frame_format

yaacov, 2018-06-24:
@avnerkhan hi, 

I added tests for the server part of this lib to test that it will answer/not answer to the currect unitID: https://github.com/yaacov/node-modbus-serial/pull/223

This has the side effect of testing that the tcp client send the currect unitID downstream. So I do not think the tcp client send the wrong id. You can check the GridConnect and the controller docs to see if you are missing something on that side.

avnerkhan, 2018-06-25:
Ok, thank you!

closed 2018-06-25 by avnerkhan