yaacov/node-modbus-serial #218: failed to install on Windows 7 32bit tablet says npm ERR! Failed at the serialport@6.2.0 install script.

V3NK47, 2018-06-23:
C:\test>npm install modbus-serial
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readlin
e
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimi
ter
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte
-length

> serialport@6.2.0 install C:\test\node_modules\serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install connect ETIMEDOUT 52.216.129.219:443

C:\test\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Pro
gram Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\.
.\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files
\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYT
HON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Program Files\nodejs\node_mo
dules\npm\node_modules\node-gyp\lib\configure.js:483:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Program Files\nodejs\node_mod
ules\npm\node_modules\node-gyp\lib\configure.js:508:16)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\grac
eful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodej
s\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\test\node_modules\serialport
gyp ERR! node -v v8.11.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open 'C:\test\package.json'
npm WARN test No description
npm WARN test No repository field.
npm WARN test No README data
npm WARN test No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
**npm ERR! Failed at the serialport@6.2.0 install script**.
npm ERR! This is probably not a problem with npm. There is likely additional log
ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\USER 2\AppData\Roaming\npm-cache\_logs\2018-06-21T08_00_25
_064Z-debug.log

C:\test>npm install modbus-serial --unsafe-perm --build-from-source
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte
-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readlin
e
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimi
ter

> serialport@6.2.0 install C:\test\node_modules\serialport
> prebuild-install || node-gyp rebuild

C:\test\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Pro
gram Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\.
.\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files
\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYT
HON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Program Files\nodejs\node_mo
dules\npm\node_modules\node-gyp\lib\configure.js:483:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Program Files\nodejs\node_mod
ules\npm\node_modules\node-gyp\lib\configure.js:508:16)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\grac
eful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodej
s\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\test\node_modules\serialport
gyp ERR! node -v v8.11.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open 'C:\test\package.json'
npm WARN test No description
npm WARN test No repository field.
npm WARN test No README data
npm WARN test No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional log
ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\USER 2\AppData\Roaming\npm-cache\_logs\2018-06-21T08_01_04
_907Z-debug.log

**_Attached Logs_**
[2018-06-21T08_00_25_064Z-debug.log](https://github.com/yaacov/node-modbus-serial/files/2129763/2018-06-21T08_00_25_064Z-debug.log)
[2018-06-21T08_01_04_907Z-debug.log](https://github.com/yaacov/node-modbus-serial/files/2129764/2018-06-21T08_01_04_907Z-debug.log)

yaacov, 2018-06-23:
Hi, Thanks for the issue,

can you remove the dependency on `serialport` here:

https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

and try again, if you do not need serial communication (e.g. tablet with no serial port) it should work.

see: https://github.com/yaacov/node-modbus-serial/issues/88

V3NK47, 2018-06-23:
I am using Panasonic ToughPad Fz G1. It has a USB 3.0 port and I'm using ModbusRTU as the mode for communication.

yaacov, 2018-06-23:
From the logs it looks like `node-serialport` fails to install, so first, you will need to install it, take a look here:
https://github.com/node-serialport/node-serialport#installation-instructions
and try to see if you can make `node-serialport` to work.

yaacov, 2018-06-24:
@V3NK47 from your errors, you are missing:

https://github.com/nodejs/node-gyp#on-windows
https://github.com/nodejs/node-gyp#installation

V3NK47, 2018-06-25:
Thanks for the solution. Let me try and get back.

dushyantbangal, 2018-06-25:
I've faced installation issues with `serialport` on Linux and Windows (even without `npm install modbus-serial`). Looks like you haven't installed python or don't have the `PATH` set to its location.

Installing `node-gyp` properly and then `npm install serialport --build-from-source` always worked for me.

V3NK47, 2018-06-26:
Yes Worked very well. Thank you for the support.

closed 2018-06-26 by V3NK47