yaacov/node-modbus-serial #224: Timeout Error object does not contain code or errno

dushyantbangal, 2018-06-26:
I am not getting an object containing `code` or `errno` in case of timeout.

When I `console.log(err`) in case of timeout, heres what I get:
```
Error: Timed out
    at Timeout._onTimeout (X:\myproject\node_modules\modbus-serial\index.js:172:30)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
```

In this case, both `err.code` and `err.errno` are `undefined`, however, the examples suggest that they should be `ETIMEDOUT`

In case when the server is not available, following is the output of `console.log(err)`
```
{ Error: connect ECONNREFUSED 127.0.0.1:502
    at Object._errnoException (util.js:992:11)
    at _exceptionWithHostPort (util.js:1014:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1186:14)
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect',
  address: '127.0.0.1',
  port: 502 }
```

I believe, timeout error is supposed to have a similar structure.

dushyantbangal, 2018-06-26:
I think it is giving out `err.message` along with stacktrace instead of the whole `err` object.

yaacov, 2018-06-26:
Thanks :+1: 

I hope to have time to look at it this weekend, if you find a fix before I find one, please open a pull request.

dushyantbangal, 2018-06-26:
I was thinking about the same thing. Can you just point me where I should look? I couldn't find `ETIMEDOUT` anywhere in the code.

yaacov, 2018-06-26:
> Can you just point me where I should look?

The general direction is here :-)

https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L46

closed 2018-06-26 by yaacov

dushyantbangal, 2018-06-27:
> The general direction is here :-)
I think that is about the connection request timed out; my issue was about transaction timed out.
But looks like both are related as the `timeout` event is sending error without `code` or `errno`
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L98

commit 08999c56 (2018-06-26): Fixes #224
Added object for ETIMEDOUT
commit 27aaf780 (2018-06-26): Merge pull request #225 from dushyantbangal/master

Fixes #224