yaacov/node-modbus-serial #226: Continue #210

yaacov, 2018-06-27:
Continue discussion from https://github.com/yaacov/node-modbus-serial/issues/210

cc// @pkaz @dushyantbangal @debevv

**Issue**:
a. a sample client would go like the code below.
b. It connects to a Serial over TCP/IP (encapsulated) device.
c. If I try to read Input Registers on multiple requests, it always fails on the second and never answers.

**Sample client**:
``` javascript
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
client.connectTelnet("10.10.10.159", {port:4001});
client.setID(1);

setInterval(async function() {

  let results=[];
  results.push(await client.readInputRegisters(0, 12));
  results.push(await client.readInputRegisters(20, 2));
  results.push(await client.readInputRegisters(28, 12));
  results.push(await client.readInputRegisters(46, 2));
  // console.log(results);

  results.forEach(result => {
    for (let i=0; i<result.buffer.length; i+=4){
      console.log(result.buffer.readFloatLE(i));
    }
  });

}, 500);
```

@debevv
I will pull the new version that got merged and try again later today. But as I said, mine was a client side problem...

@All_of_us
I have found out that if I try to Read more than 21 Registers there is an exception as well, that is why I went with the multiple request solution with await. @dushyantbangal is right, they are not parallel request... I would like to get the full length of registers in one request.

yaacov, 2018-06-27:
Coping  @dushyantbangal  commant https://github.com/yaacov/node-modbus-serial/issues/210#issuecomment-400544602

@pkaz, It might be possible that whichever device you are reading from does not allow more than 20 registers in one request. I have come across some meters that limited reading 40 registers in a single request. So I dont think that is fixable from this side, its the manufacturer.

Regarding the parallel issue, it is possible that your TCP-RTU gateway is messing up as thinking its all a single request (happened to me). Try putting a small delay between 2 reads - like 50 ms. They still will be kind-of parallel requests as you are not waiting for one to complete before firing the next one.

Try client.connectTcpRTUBuffered instead of client.connectTelnet if possible.

dushyantbangal, 2018-06-28:
> @dushyantbangal Tried it (connectTcpRTUBuffered), the promise never responds... :( I think I am doing something wrong...

@pkaz, If you have tried `client.connectTcpRTUBuffered`, then I think it might be issue from the TCP-RTU gateway you are using. I would suggest you try it out with a simulator or the `server.js` from the examples.

If everything is working with simulator, then I would suggest the following from my previous comment:

> Try putting a small delay between 2 reads - like 50 ms. They still will be kind-of parallel requests as you are not waiting for one to complete before firing the next one.

pkaz, 2018-06-28:
@dushyantbangal So, I have a ElControl - STAR3 Energy Analyzer behind a MOXA serial2Ethernet gateway. I am not using a simulator, but have an actual device.

The problem is I am kind of new to the field of SCADA and protocols, so I am trying to figure things out as I go. I will try my best to setup a simulation, but it will never be the real thing. So as far as I can I will try to test the client side of things with the actual device working as a service. Is that ok? Can I help any other way?

yaacov, 2018-07-06:
Hi all,

Just noticed something, if the controller is connected using serial line (at some point of the way) you should not use parallel requests.

For example, if you have a controller connected via rs485 to some TCP-to-485 converter, even if you use the tcp protocol, you can not do parallel calls, they will just get stuck when the converter try to push them using the rs485 line, because most converters will not buffer messages :-(

dushyantbangal, 2018-07-06:
@yaacov, thats exactly what I had faced once. I kept getting the "expected 2 but got 1" kind of error, and only used to get one of the responses.
I just added a small delay between multiple requests, and it worked. You don't have to wait for the response, but just send the requests with a small delay.
However, this will still depend on how the converters are handling it. There will be some trial-and-error involved.

yaacov, 2019-01-19:
I assume the problem was "the controller is connected using serial line (at some point of the way) you should not use parallel requests.", and once all requests are done serially, the problem is fixed ?

Closing for now,

closed 2019-01-19 by yaacov