yaacov/node-modbus-serial #227: Exception Code #1

avnerkhan, 2018-07-06:
Hello , 

I am reading from two types of devices , RS-485 and DIO. They are connected to my server through a hub and they have seperate IPs. I can read either the DIO IP or the RS-485 IP fine in my program, but issues arise when I try to read them both at the same time. Currently, I am putting all the devices that read the RS-485 in one map and all the DIO in another map. I then proceed to read through the DIO first and then RS-485. I made sure that there is enough timing to open up the ports when switching IPs .I read the RS-485 devices fine, but I cannot read the DIO devices, even though I could do it in a isolated setting. I keep getting modbus exception code #1, which means an illegal function, but this function worked in the isoalted setting. I was wondering under what other circumstances this issue might arise.

yaacov, 2018-07-07:
do some devices of different protocol share the same physical rs485 line at some point ?

avnerkhan, 2018-07-07:
Well my laptop is connected to the hub via ethernet. The hub connects both a DIO reader and a RS-485 reader (They both should have modbus capability)with two seperate ethernets. As far as that goes, that is the extent of the connectivity between the two protocols. Also, it might help to add that I am using the "readDiscreteInput" method on the DIO to read the input values, because my DIO device does not have explicit holding registers. The parameters that I send to that method are valid and I usually get the correct response, but it gives exception code 1 when I try to read it with RS-485, as mentioned before.

yaacov, 2018-07-07:
If they have two different IP addresses then it shouldn't happen [ they should not know about each other ] :-(
When you find out what is happening, please comment here or open a pull request if it needs fixing ...

yaacov, 2018-07-08:
p.s.

your code looks like this ?

``` javascript
...
var client_mod = new ModbusRTU();
var client_dio = new ModbusRTU();

client_mod.connectTCP("some_ip")
    .then(...)
    .catch(...)

client_dio.connectTCP("some_other_ip")
    .then(...)
    .catch(...)

...
```

avnerkhan, 2018-07-09:
Works perfectly when I tried creating two clients ,as you suggested. For some reason I didn't think of doing that. Thank you!

closed 2018-07-09 by avnerkhan