yaacov/node-modbus-serial #228: no issue

vipulpatel1994, 2018-07-09:
Sorry for rising here my query not getting any option to contact you.
Where exactly we can use this.
I have one modem which can connect to tcp/ip. need to send one hex string(01 04 00 01 00 01 60 0A) to get data after connection. able to connect but not able to send back hex can you help me out.

yaacov, 2018-07-10:
Hi thanks for the issue,

This vary from device to device, you will need to consult with the modem manual, and the docs for the software you use.

vipulpatel1994, 2018-07-17:
Thanks a lot for help @dushyantbangal and @yaacov  . for now done with custom code. I will surely try this.

yaacov, 2018-08-14:
> for now done with custom code.

:smiley: 

@vipulpatel1994 closing. feel free to open a new issue if needed.

closed 2018-08-14 by yaacov