yaacov/node-modbus-serial #230: Ping rate on discreteInputs

avnerkhan, 2018-08-22:
Hello,

I have a program written that connects to various clients and pings machine connected to these clients at a certain rate. One of these clients is a DIO device in which I have to read the discrete input register (FC=02 , client.readDiscreteInput()). If I ping the machines anywhere from 1-30 seconds, the DIO device reads fine. However, if I ping at 40-60 seconds, then I receive a "Port not open" error on the DIO interface. All other interfaces (that read holding registers instead of discrete inputs) read fine on a 60 second ping rate.

Thank you, Avner Khan

yaacov, 2019-01-19:
Hi @avnerkhan, sorry for the very late response,  is this fixed ?

dushyantbangal, 2019-01-21:
@yaacov I have come across some PLCs that close the connection after n number of seconds of inactivity. The number is configurable in some cases.
@avnerkhan if by "all other interfaces" for mean the devices other than the DIO (and not holding regs on DIO), I think its an issue from the DIO side. I think your DIO device is the one that is closing the connection after 30 seconds of inactivity. Maybe they let you configure the number somewhere.

yaacov, 2019-10-12:
closing as stale

closed 2019-10-12 by yaacov