yaacov/node-modbus-serial #231: Writing to Modbus server with Buffer instead of Integer value

jangwonsuh, 2018-09-07:
Hi,

I was wondering if you can add a feature where you can use a Buffer as an argument for the writeRegisters() function instead of an integer value? I am currently storing Strings, Floats, Longs, and other data types in the Modbus server. I can easily convert them to Buffer values. however, the writeRegister() function only takes in Uint16 values and converts that back to a Buffer value, which seems repetitive. Similar to how the read request returns a Buffer value, can you also allow to write with a Buffer value? Thanks.

yaacov, 2018-09-07:
@jangwonsuh tanks for the issue.

Can you make a pull request adding that feature ?

vlastoun, 2019-04-28:
@jangwonsuh is there any progress? any way i can store float to server?

yaacov, 2019-10-12:
fixed by: #282

closed 2019-10-12 by yaacov