yaacov/node-modbus-serial #232: Problem with timeout in modbusTCP client

behi88, 2018-09-09:
hello
I have a problem with timeout.
when I add time out to my client modbus TCP no matter in the device answered or not I always receive timeout here is my code:

client.setTimeout(5000);
client.setID(1);
client.connectTCP(IP, {port: 502}).catch (function () {});
		
setInterval(function() 
{
    client.readHoldingRegisters(0, 10, function(err, data) 
	{
		if (err)
		{
			console.log('SID:('+client.getID()+") " + err.message);
		}
		else
		{
			console.log(data.data);
		}
    });
},1000);

yaacov, 2018-09-09:
@behi88 Thanks for the issue !

I will look at it, but I will not have lots of time in the next few weeks.

a. If you find the problem, please comment here with the details you find out.
b. If you find the problem and fix it, please make a pull request with the fix.

Thanks

behi88, 2018-09-09:
#yaacov I think I found the problem maybe I am wrong but I could not find Timeout Cancel for TCP or other Protocols the timeout Cancel is only available in Serial mode.

yaacov, 2018-09-09:
Thanks, some info:

We have **two** timeout:

1. for all ports , using js `setTimeout`:

https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L174
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L245

2. only for tcp, using tcp ports internal timeout:

https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L46
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L98

behi88, 2018-09-09:
I think this is it:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L46
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L98
there is no 'clearTimeout' on data receive 
it will take me a while to figure the code out and pin point where to insert it
thanks for info.

behi88, 2018-09-11:
sorry @yaacov there was a mix up with salve device and your codes are great.
please close the topic.