yaacov/node-modbus-serial #233: Using C701 UDP protocol

behi88, 2018-09-12:
Hello @yaacov
I try to use C701 Protocol but there was no documentation or example.
so actually I try to figure the thing out by looking as source comments.
I use every simulator like 'modbus toolbox' to connect to a slave as UDP.
but there was no connection. I use "wireshark"  & "Comm Operator" to analyze the packets but the connection result was not UDP it was some kind of TCP but Broken! 
I used a simple client program and just replace "client.connectTCP" with "client.connectC701".
could you please help me on this?

yaacov, 2018-09-12:
@behi88 hi, C701 is a proprietary protocol, and currently I have no machine that speaks it, to check if it works or not ... 

It was built for specific use case and since I have no issue on that specific machine I do not want to change it.

What type of machine you want to connect to ?

behi88, 2018-09-12:
I use RPI as master and a 'modbus toolbox' software on windows PC.
I also try to analyze packets with "wireshark" & "Comm Operator" software.

yaacov, 2018-09-12:
C701 is not exactly modbus (it's a proprietary dialect), so it should not connect to a regular modbus client ...

behi88, 2018-09-12:
is the protocol UDP? am I using it right? I just replace .connectTCP with .connectC701. do I have to add anything else?

yaacov, 2018-09-12:
I do not remember, it was a lot of time ago ... :-)

behi88, 2018-09-12:
OK thanks Great code bytheway.

closed 2018-09-12 by behi88