yaacov/node-modbus-serial #234: Error: read/write function putted into an Infinite Loop

lphquy, 2018-09-13:
here is my code to read a discrete input by ModbusRTU: 

/* ============================================*/
function read_DI1() {
    client.setID(1);
    console.log('setID completed!');
    client.readDiscreteInputs(0, 1, function(err, data) {
        if (err) throw err;
        else {
            var DI1_callback = data.buffer.readUInt16BE();
            console.log(DI1_callback);
        }
    });
};

client.connectRTU("/dev/ttyAMA0", {baudRate: 9600}, function(err, data) {
    if (err) {
        throw err;
    } else {
        console.log('port opened!');
        read_DO1();
    }
});
/* ============================================*/

So, my program will putted into an Infinite Loop and I can't print the DI1_callback value through the command line. 
[image]

I'm extremely thank you for your helping.

yaacov, 2018-09-13:
Hi, thanks for the issue.

Can you try something more full, like the examples in :
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js

And see if you get more info ?

lphquy, 2018-09-14:
Hi, thanks yaacov so much! 
My code is out of loop because of the "client.close()". So thank you, but, I still faced with the another stuff:
"The timeout error" 
[image]

here is my code:

//=================================================//
var ModbusRTU = require("./node_modules/modbus-serial/index");
var client = new ModbusRTU();
var GPIO = require('onoff').Gpio;
var pin15 = new GPIO(22, 'out');       //GPIO22 = Pin15 output
var pin13 = new GPIO(27, 'out');        //GPIO27 = Pin13 output

/**Pin13 = 0 => choose GSM  */
/**Pin13 = 1 => choose Modbus */
/**Pin15 = 0 => choose read  */
/**Pin15 = 1 => choose write */

var DO1Value = true;
var pin13Value = 1; //modbus
var pin15Read = 0;
var pin15Write = 1;

var networkErrors = ["ESOCKETTIMEDOUT", "ETIMEDOUT", "ECONNRESET", "ECONNREFUSED", "EHOSTUNREACH"];

/** write DO1 */
function read_DO1() {
    pin15.writeSync(pin15Write);
    console.log('set pin15Write completed!');
    pin13.writeSync(pin13Value);
    console.log('set pin13 Modbus completed!');

    client.setID(1);
    console.log('setID completed!');
    client.setTimeout(1000);

};

client.connectRTU("/dev/ttyAMA0", { hupcl: false, dsrdtr: false })
    .then(setClient)
        .then(function() {
            console.log("Port Opened"); })
        .catch(function(e) {
            if(e.errno) {
                if(networkErrors.includes(e.errno)) {
                    console.log("we have to reconnect");
                }
            }
            console.log(e.message);
        });

function setClient() {
    client.setID(1);
    client.setTimeout(2000);
    readDI();
}

function readDI() {
    client.readDiscreteInputs(2, 1)
        .then(function (data) {
            console.log("Received: ", data.data); })
        .catch(function (err) {
            console.log("Error: ", err.message); })
        .then(close);
}

function close() {
    client.close();
}
//==============================// 

Can you have any ideas about my timeout error, yaacov? Thank you so much more!!!

yaacov, 2018-09-14:
It looks like the device does not answer after timeout ( you set it to 2sec )...
or it may answer with corrupt data ...

a. try removing/changing timeouts -
see line `client.setTimeout(2000);`

b. if you can listen to the line, try to understand what is sent to device and if device answer.

yaacov, 2018-09-14:
a. You can try to use the debug mode, to see if the device returns something by setting this environment variable [1]:
```
export DEBUG=modbus-serial
```

b. You should also check your wiring / settings  / jumpers on the device ... maybe it has some settings to not answer or it need some jumper to be in some setting.

c. If you have some connection RS485 <-> RS232 or <-> usb you should also check you use it currectly.

[1] https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

closed 2018-10-15 by lphquy