yaacov/node-modbus-serial #240: read/write fail after reconnecting RTU

dushyantbangal, 2018-10-18:
I am using a RS485-USB converter to connect to meter.
I am on Windows 10, node 8, latest package version from npm.

I have a simple write command followed by a read command, both wrapped in a `setInterval`
Before the commands, I check if the port if open or not. If its not, then I start reconnection. The connection is retried every 1500 ms.

If I unplug the converter and plug it back once it starts reconnection, the connection is successful. However, most of the times, the requests keep failing, until I kill the script and restart it again.

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
var interval;

connect();

function write() {

    if(!client.isOpen)
    {
        clearInterval(interval);
        connect();
        return;
    }

    console.log("sending write");
    client.setID(1);

    client.writeRegister(5, 2).then(function(resp){
        console.log(resp);
        read();
    }).catch(function(err){
        console.log(err);
    })
        
}

function read() {
    console.log("sending read");

    client.readHoldingRegisters(5, 1)
        .then(console.log)
        .catch(console.log);
}

function connect(){
    client.connectRTU("COM5", { baudRate: 9600, parity: 'none', dataBits:8, stopBits:1 }, function(err){
        if(err){
            console.log(err);
            return setTimeout(connect,1500);
        }else{
            console.log("connected");
            interval = setInterval(write, 5000);
        }
    });
}
```

Output:

```
connected
sending write
{ address: 5, value: 2 }
sending read
{ data: [ 2 ], buffer: <Buffer 00 02> }
sending write
{ address: 5, value: 2 }
sending read
{ data: [ 2 ], buffer: <Buffer 00 02> }        //UNPLUG
Error: Opening COM5: File not found
Error: Opening COM5: File not found
Error: Opening COM5: File not found            //PLUG
connected
sending write
{ address: 5, value: 2 }
sending read
sending write
{ address: 5, value: 2 }
sending read
sending write
{ address: 5, value: 2 }
sending read
```

yaacov, 2018-10-18:
@dushyantbangal Thanks :+1: , I hope to have time to look at it this weekend, but I have very little time ...
If you can investigate more / find a solution it will be great.

gabrielpereirapinheiro, 2018-12-10:
I'm having the same problem ! Someone find the soluction and can asnwer here  ?!?

gabrielpereirapinheiro, 2018-12-10:
I'm having the same problem !
Someone find the solution and can answer here ? Thanks

yaacov, 2018-12-10:
@dushyantbangal  @gabrielpereirapinheiro hi, sorry for not adapting ... 

p.s.

looking at the example code from https://github.com/yaacov/node-modbus-serial/issues/240#issue-371407616

can you do a
`setTimeout(write,5000);`
in the end of `connect` and in the end of `read`
instead of one:
`setInterval(write, 5000);`
in the end of `connect`

It will not fix the problem ... but one it will be one less `setInterval` to worry about :-)

p.s.
This may be a windows specific problem @gabrielpereirapinheiro do you use windows too ?

gabrielpereirapinheiro, 2018-12-14:
At really my problem is :  I'm using the library in raspberry pi to read data from a sensor and always the program stop reading in different times after start.

sigma-design, 2019-01-24:
Hi @gabrielpereirapinheiro ,

Recently I was implementing polling Modbus RTU units via Modbus TCP gateway and came up with the framework that seems to be working for this case as well, so maybe it can help you.

Please find below this framework that I have adapted for RTU and tried with USB-to-RS232 converter (plugging/unplugging several times):

```
"use strict"

//==============================================================
// create an empty modbus client
var ModbusRTU   = require ('modbus-serial');
var client      = new ModbusRTU();

var mbsStatus   = 'Initializing...';    // holds a status of Modbus

// Modbus 'state' constants
var MBS_STATE_INIT          = 'State init';
var MBS_STATE_IDLE          = 'State idle';
var MBS_STATE_NEXT          = 'State next';
var MBS_STATE_GOOD_READ     = 'State good (read)';
var MBS_STATE_FAIL_READ     = 'State fail (read)';
var MBS_STATE_GOOD_CONNECT  = 'State good (port)';
var MBS_STATE_FAIL_CONNECT  = 'State fail (port)';

// Modbus configuration values
var mbsId       = 1;
var mbsScan     = 1000;
var mbsTimeout  = 5000;
var mbsState    = MBS_STATE_INIT;
var mbsBufData;

//==============================================================
var connectClient = function()
{
    // set requests parameters
    client.setID      (mbsId);
    client.setTimeout (mbsTimeout);

    // try to connect
    client.connectRTUBuffered ("COM9", { baudRate: 9600, parity: 'even', dataBits:8, stopBits:1 })
        .then(function()
        {
            mbsState  = MBS_STATE_GOOD_CONNECT;
            mbsStatus = 'Connected, wait for reading...';
            console.log(mbsStatus);
        })
        .catch(function(e)
        {
            mbsState  = MBS_STATE_FAIL_CONNECT;
            mbsStatus = e.message;
            console.log(e);
        });
}

//==============================================================
var readModbusData = function()
{
    // try to read data
    client.readHoldingRegisters (5, 1)
        .then(function(data)
        {
            mbsState   = MBS_STATE_GOOD_READ;
            mbsStatus  = 'success';
            mbsBufData = data.buffer.swap16();
            console.log(data.buffer);
        })
        .catch(function(e)
        {
            mbsState  = MBS_STATE_FAIL_READ;
            mbsStatus = e.message;
            console.log(e);
        });
}

//==============================================================
var runModbus = function()
{
    var nextAction;

    switch (mbsState)
    {
        case MBS_STATE_INIT:
            nextAction = connectClient;
            break;

        case MBS_STATE_NEXT:
            nextAction = readModbusData;
            break;

        case MBS_STATE_GOOD_CONNECT:
            nextAction = readModbusData;
            break;

        case MBS_STATE_FAIL_CONNECT:
            nextAction = connectClient;
            break;

        case MBS_STATE_GOOD_READ:
            nextAction = readModbusData;
            break;

        case MBS_STATE_FAIL_READ:
            if (client.isOpen)  { mbsState = MBS_STATE_NEXT;  }
            else                { nextAction = connectClient; }
            break;

        default:
    }

    console.log();
    console.log(nextAction);

    // execute "next action" function if defined
    if (nextAction != undefined)
    {
        nextAction();
        mbsState = MBS_STATE_IDLE;
    }

    // set for next run
    setTimeout (runModbus, mbsScan);
}

//==============================================================
runModbus();

```

yaacov, 2019-01-24:
@sigma-design can you add your code to the examples ?

closed 2019-10-12 by yaacov