yaacov/node-modbus-serial #248: How to catch CRC errors

buuuudzik, 2019-01-15:
Hello,

I have from time to time such error:
Error: CRC error
    at RTUBufferedPort.<anonymous> (C:\Users\me\Desktop\inverter\node_modules\modbus-serial\index.js:274:42)
    at RTUBufferedPort.emit (events.js:182:13)
    at RTUBufferedPort._emitData (C:\Users\me\Desktop\inverter\node_modules\modbus-serial\ports\rtubufferedport.js:108:10)
    at SerialPort.onData (C:\Users\me\Desktop\inverter\node_modules\modbus-serial\ports\rtubufferedport.js:67:22)
    at SerialPort.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
    at SerialPort.Readable.push (_stream_readable.js:219:10)
    at binding.read.then (C:\Users\me\Desktop\inverter\node_modules\serialport\lib\serialport.js:374:10)

How can I catch it?

I am trying to log errors but error is from library, not as a result of reading:
`client.readHoldingRegisters(address, length, (err, res) => {
if (err) return console.log(err); // error isn't logged
});`

yaacov, 2019-01-15:
Hi, thank you for the issue :100: 

a - can tou try to call `writeFC3` directly `readHoldingRegisters` is a syntactic sugar around it, see:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L467 ?

b - can you check if this issue exist also in the promise systex, e.g. using `.then`, `.catch`,
see for example, here: https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js#L14 ?

If you find what is the problem and can fix it, please comment here.

Thanks :+1:

buuuudzik, 2019-01-15:
Now I am still testing. 

2 hours ago I've tried use then().catch(). It logged no error and whole script works without any error. But it is still very dangerous. I must handle this kind of error to know that my solution is error resist;)

This was my change:
`
// old version
const runner = (address, length, callback) => client.readHoldingRegisters(address, length, callback);

        // new version
        const runner = (address, length, callback) => {
            client.readHoldingRegisters(address, length).then(res => {
                if (callback) callback(false, res);
            }).catch(err => {
                console.log('Error catched!', err);
                if (callback) callback(err, false);
            });
        };
`

yaacov, 2019-01-15:
a - did you check with `writeFC3` ?

b - 

> still very dangerous

you can check that the `data` payload is valid, if an error happened and you got into `.then` instead of `.catch` you will get none valid data.

p.s.
I looked at the code ( an error prof way to find bugs :-) ) and can not see the where we miss catching the error :-( , if you can find the point where it happen it will be very helpful.

buuuudzik, 2019-01-15:
I cannot reproduce now this issue. I think this could be caused by some noise from cable and now maybe there is no noice because whole devices are switched off in the office. I will figure it out;)

yaacov, 2019-06-19:
closing for now, as this is not reproducible now

feel free to re-open or open a new issue when we see this again, and can debug ...

closed 2019-06-19 by yaacov

dushyantbangal, 2020-06-19:
I have faced this issue recently. It is mostly because of noise. I solved it by closing my connection from time to time.
Initially, I kept the connection open continuously, and after some time, I used to get a CRC error, and then no responses for any of the commands after that. I debugged using the `DEBUG` variable, and saw I was getting a lot of garbage.

**Opening connection before read, closing after solved it.**