yaacov/node-modbus-serial #249: serialport an optional dependency, take 2

emes, 2019-01-19:
Would you re-consider making serialport an optional dependency? Docker images become muuuch larger when serialport needs to be built from source, which is shame when using only tcp.

yaacov, 2019-01-19:
Or maybe having two npm packages is a good solution ?!

yaacov, 2019-01-19:
and we can remove the debug dependency too ...

yaacov, 2019-01-19:
I made a version without serialport :tada: :balloon: :cake: 

https://www.npmjs.com/package/modbus-serial/v/7.5.1-no-serial-port.

Note: I will now have to remember to make a `no-serial-port` versions ... will see how that goes :-)

emes, 2019-01-19:
wow.  that's awesome.  modbus-noserial! 🎂 😄

yaacov, 2019-01-19:
closed by #250

closed 2019-01-19 by yaacov

Craigzyc, 2023-05-02:
For anyone else landing here with the same issue I was having. 
The 7.5.1-no-serial-port version released by yaacov still had references to serialport which caused it to fail 
Over at https://github.com/Craigzyc/node-modbus-serial.git I have removed all the serialport references on the current version (8.0.11) if you need it

The serial port requirement was causing issues trying to use the package for pure TCP in docker. I spent the better part of a morning trying to track down the issues before making this solution. Hopefully it saves someone some time