yaacov/node-modbus-serial #251: Is it possible to keep TCP connection open on remote unit response timeout ?

sigma-design, 2019-01-22:
Hi @yaacov,

Thanks for this package, it helps in establishing remote access to units via Modbus TCP in my case.

So far everything is good, but it seems that the TCP connection is closed on Modbus response timeout while remote TCP/IP is still available - in my case the remote Modbus RTU unit is connected to Modbus TCP gateway via serial port and such timeouts will occur when Modbus RTU unit is disconnected (or powered down) while Modbus TCP gateway (separate physical device) is still up and running.

This requires reconnecting TCP for the next request which seems to be unnecessary, so is it possible not to close TCP connection explicitly on Modbus response timeout exception ?

N.B. For example, commercial Modbus Poll software keeps the same TCP connection open in this situation.

Regards

yaacov, 2019-01-22:
Hi, thanks for the issue !

You can try and see what happen with different timeout settings.

ths timeout for tcp (non buffered) is set here:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L46
bufferd is set here:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcprtubufferedport.js#L53

net socket timeout docs:
https://nodejs.org/api/net.html#net_socket_settimeout_timeout_callback

yaacov, 2019-01-22:
:+1: 

We call the `timeout` event here:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcprtubufferedport.js#L124

yaacov, 2019-01-22:
> but at the moment I am not sure how it could be implemented nicely in the core code.

:+1:  Thanks, that makes sense, 

What happen if you set the `timeout` to be never ? does that solve the problem on your end ?

p.s.
Feel free to test things :-) if you think you have a nice ( or even just something that will fix you issue ) please make a pull request.

sigma-design, 2019-01-22:
Having "no timeout" is not an option for me as I need a re-try mechanism for robustness and feedback, so I did not check it out.

After further thinking/testing, it looks like that the issue can be resolved by simply removing the "modbus.openFlag = false;" code line in the 'timeout' event handlers (tcpport.js and tcprtubufferedport.js).

The rationale is that the state of this flag in the timeout event handler actually reflects two types of timeouts, i.e. for "TCP connection timeout" its value is 'false' while for "Modbus response timeout" its value is 'true' - this is expected behaviour and it seems that leaving the flag in its state is the appropriate way to go.

I have tested the above solution and it worked fine for me, i.e. "TCP connection timeout" event kept port closed and "Modbus response timeout" event kept port open, so the Modbus reading re-tries can be executed without TCP reconnection.

If you agree with the above solution, maybe you can update the code by just removing these lines (probably adding comment about rationale), alternatively I can make a pull request if you prefer so.

Regards

yaacov, 2019-01-23:
> If you agree with the above solution, maybe you can update the code by just removing these lines (probably adding comment about rationale), alternatively I can make a pull request if you prefer so.

:+1:

Thanks, pull request is preferred, with ref to this issue.

yaacov, 2019-01-23:
closed by #252

closed 2019-01-23 by yaacov