yaacov/node-modbus-serial #253: Queue? Has anyone succesfully implemented a queue for the BufferedRTU Connection

RobertSmart, 2019-01-23:


yaacov, 2019-01-23:
a - You can try also asking in https://github.com/biancode/node-red-contrib-modbus
b - It may be something we can fix, if you see a way to fix it please make a pull request, it may help other people too.

cc @biancode

Juanmabs22, 2019-06-19:
Yes, I find the way!

You need a way to have a mutex, semaphore or something to make your RTU only running at one time.
[You need a async queue.](http://caolan.github.io/async/v3/docs.html#queue)

Then, create a queue with only 1 worker, if there only one worker using the rtu none else will use it => Safethread! noCRCproblems!

```
const queue = require('async/queue');
let queueRTU = queue(this.runner.bind(this), 1);
```

then, create the function to handle the jobs on the worker:

    async runner(thingToProcess) {
        [Here you will use the rtu]
    }

To do it simple, the runner will receive a promise+action, so the runner will run the action and resolve the promise when ends, the runner won't know whats is doing, but we want only that the worker make only one thing at time:

    async runner(magicPromise) {
        try{
            magicPromise.resolve(await magicPromise.action());
        } catch (error) {
            magicPromise.reject(error);
        }
        // callback(); // You won't use callback due the async enviroment
    }

next, you will create the jobs to run the worker, I mean, the magicPromise:

        const runInSafeThread = (action) => {
            let resolve, reject;
            let magicPromise = new Promise((res, rej) => {
                resolve = res;
                reject = rej;
            });
            magicPromise.action= action;
            magicPromise.resolve = resolve;
            magicPromise.reject  = reject;

            queueRTU.push(magicPromise);

            return magicPromise;
        };
Tha magicPromises combines a Promise with a function to send all together to the worker to the worker would be able to run the action and notify us when finish the action.

You have now the environment!
Now you only have to send actions.

    let result = await runInSafeThread (async () => await rtuClientIniciazated.readInputRegisters(14, 1));
    console.log(result);

Check that we await 2 times:

- the awaited readInputRegisters is to return the data read
- the first await is to wait until the worker run the action send and get the result.

It a bit difficult, but if you like async/await you could love this approach.

Hope it helps :)
_Sorry for be too late_

yaacov, 2019-06-19:
Great ! thanks for sharing :heart: 

p.s.
Can make a PR that adds an example ?

Juanmabs22, 2019-06-19:
Just added here:
https://github.com/yaacov/node-modbus-serial/wiki/Example:--server

I'm not sure if it's ok... I usually not use Github 😁

yaacov, 2019-06-19:
> I'm not sure if it's ok... I usually not use Github

It's fine, thanks :+1:

p.s.
If you want to add an example here:
https://github.com/yaacov/node-modbus-serial/tree/master/examples
And want help about that, comment here :-)

yaacov, 2019-06-19:
closing per wiki edit.

closed 2019-06-19 by yaacov