yaacov/node-modbus-serial #257: Got error code EMFILE for connect command

dushyantbangal, 2019-02-11:
I'm using Raspberry Pi + Raspbian OS.
Every minute, my script connects to few IPs, pulls some data, and then closes conenctions.
My script was running fine for hours/days.

Then suddenly I started getting `EMFILE` error code, in the `connectTCP` callback.

I am doing some file reading, but thats only at the start of script.

And there are some other services running that do some file reads/writes.

But I don't understand why `connectTCP` will throw that kind of error.
After I killed and restarted the script, it started working fine.

yaacov, 2019-02-11:
Hi, Thanks for the issue.

If you/someone else find some more info about this, please post it here.

dushyantbangal, 2019-08-16:
Got this again. I think its happening after the connections keep failing for a long time.

yaacov, 2019-08-16:
Hi,
in https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js we have a mechanism to recover from errors, can you check if this mechanism can help in your case ?

dushyantbangal, 2019-08-16:
@yaacov hey, I've already implemented that. This error is when I create a new client and fire the `connectTCP`.
Every minute, I create the a new client, connect, read, close connection.

yaacov, 2019-10-12:
close as stale

closed 2019-10-12 by yaacov