yaacov/node-modbus-serial #258: User Defined Function Codes

apb2019, 2019-02-21:
Could you add the ability to handle User Defined Function Codes (i.e. FC 65 to 72 and 100 to 110 decimal) as I would like to be able to send commands to a solar battery system that uses FC 66 (0x42) to control functions

yaacov, 2019-02-21:
Hi,

Thanks for the issue :peach: , I do not implement new features at the moment due to lack of time :-(
But I do accept fixes and new features in the form of pull requests :exclamation: 

If you ( or someone you know ) want to implement this feature, and you wish to add it to the lib, please submit a pull request and link it to this issue.

p.s.
I also labelled this wit a `help wanted` label in case someone else will see it and want to help :heart: 

p.p.s
I will try to do it myself if I have the time, but don't wait for it ....

apb2019, 2019-02-22:
Tx, unfortunately I don’t have the programming skills to do this myself. Hopefully someone may be able to take this on as I have yet to find any Modbus software that can handle user defined codes.

yaacov, 2019-10-12:
Will be happy to get a pull request for that some day ... closing for now as stale.

closed 2019-10-12 by yaacov