yaacov/node-modbus-serial #259: getID(#) returns always one.

fedesc, 2019-02-22:
I'm trying to create a loop to find unknown addresses connected in a line of controllers.
i though sending a getID(loop_int) and see the response i get from each address.
but i always get true (1) in response. 

for an address i know exists for sure and also an adress i know doesn't exists.

i'm trying to call the function in the callback after connection was made (modbusTRUBuffered fn)

am i doing this wrong or just didn't understand ? 

Shabbat shalom ahi.

fedesc, 2019-02-22:
Now after digging a bit deeper in code i see that getID only returns this._unitID which i guess is the master Slave? i'm not sure. 
`// set/get unitID
    cl.setID = function(id) {this._unitID = id;};
    cl.getID = function() {return this._unitID;};`

Is there a way doing what i'm trying to do if not with getID? 
another approach maybe? 

thanks in advance

fedesc, 2019-02-22:
nevermind got it 🗡

yaacov, 2019-02-22:
Hi, did you find out how to do it ?

If it works for you now, please comment on the way you solved it, and close this issue.
If you didn't solve the problem, just found a work around it's also OK, just leave a note here ( and don't close the issue :-) )

yaacov, 2019-02-22:
:candle: :candle: :wine_glass: :bread:

fedesc, 2019-02-22:
It's kind of a workaround which suit to my use-case; let me explain. 

i'm building a gateway app to read a line of temperature controllers and broadcast through web.

the controllers are lined with rs485 and are identical so i use a random register which i know for a fact that exists on them and try to reach him on every address.

so... i've noticed that when i try to setID and look for a register in an address that doesn't exists, without a timeout it just hangs, unlike when the address does exists , the response is quite fast. 

So the workaround just loops a var through 0-255 and tries to read that register from each controller connected to line. if i put a decent timout i can assume that those hunged addresses don't exist and those i get data clearly do. 

So i used a recursive function in order to be able to queue each address call, one after the other and read all the range requested. 

it looks something like this:

`const ModbusRTU = require("modbus-serial");
const client = new ModbusRTU();

client.connectRTUBuffered('COM3', { baudRate: 9600 }, () => { // my line is connected on PORT COM3  >> connect
  client.setTimeout(500); // settimeout for requests - i think 500ms is a decent time to wait for a response on a connected device 
  searchSlaves(1); // start recursive function with start of range (1) 
});

const searchSlaves = (i) => {
  client.setID(i); // set address
  client.readHoldingRegisters(0,1) // try to read a known register that exists on each controller
  .then(data => { // if data exists outputs success
    console.log(`Hurray!! Slave address #${i} is connected with master.`);
    if (i<255) { // this is the loop part that goes through range recursively
      i++;
      searchSlaves(i); 
    }
  }).catch(err => { // if no response and timeout kicks in script assumes no slave exists on this address.
    console.log(`address #${i} is blank! ${err.name} -> Means no slave was found on this address`)
    if (i<255) { // this is the loop part that goes through range recursively 
      i++;
      searchSlaves(i); 
    }
  });
}`

fedesc, 2019-02-22:
this is the output (i have only 137 connected ... i did the whole range from 0-255 but pasting only the results that matters :) 
[image]

yaacov, 2019-02-22:
for code use 3 squats  ` ``` `  like this:

``` javascript
  // my line is connected on PORT COM3 >> connect
  client.connectRTUBuffered('COM3', { baudRate: 9600 }, () => {
  // settimeout for requests - i think 500ms is a decent time to wait for a response on a connected device
  client.setTimeout(500);
  // start recursive function with start of range (1)
  searchSlaves(1);
});

const searchSlaves = (i) => {
  // set address
  client.setID(i);

  // try to read a known register that exists on each controller
  client.readHoldingRegisters(0,1)
    .then(data => {
      // if data exists outputs success
      console.log(`Hurray!! Slave address #${i} is connected with master.`);
      // this is the loop part that goes through range recursively
      if (i<255) {
        i++;
       searchSlaves(i);
      }
      // if no response and timeout kicks in script assumes no slave exists on this address.
    }).catch(err => {
      console.log(`address #${i} is blank! ${err.name} -> Means no slave was found on this address`);
      // this is the loop part that goes through range recursively
      if (i<255) {
        i++;
       searchSlaves(i);
      }
    });
}
```
> i would appreciate other more efficient ways to do it :)

IMHO this is the best way ...

p.s.
I think using `finally` [1] for the recursion check is nicer :-)

[1] https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise/finally

closed 2019-02-22 by yaacov

yaacov, 2019-02-22:
p.p.s 

Most rs485 lines are limited to 32 units max, maybe you can limit the search to 32 instead of 255 ?
 This depends on the way the ID's are allocated ...