yaacov/node-modbus-serial #261: Error: Modbus exception 2

raviteja2503, 2019-03-26:
Hi I am using a RS485-USB converter to connect to an energy meter Modbus RS485 to read the modbus registers. I am trying read holding register 3203 with the help of this module but i am unable to read, i am getting the following error.
[image]

And my code snippet is:
[image]
 
I am on my windows 10 machine and modules used
Node: "10.5.0,"
modbus-serial: "7.5.1",
serialport: "7.1.4"

I tried this on my Linux machine also but no luck, Please help me in resolving the issue.
Thanks in advance for the support.

yaacov, 2019-03-26:
Hi, thanks for the issue !

a - can you try one of the examples,
For example, replace the connection and reading info, to work with your machine, in `loggrer.js`:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js

b- sometimes when device docs say register 3xxx (e.g. 3203) they mean "Analog Input Registers" number 203, you can try also reading register 203 and also trying read input registers.

raviteja2503, 2019-03-27:
Hi, Thanks for the reply.
a. I tried the examples also but no change.
b. Actually the my register when i connect from ModScan Software is 43203 and as 4 is holding register i am trying to read 3203. Even i tried 43203 also but i am facing same issue.

Here the thing is when i try to read the holding registers 0-30 i am successfully reading but more than that i am unable to do.

yaacov, 2019-03-27:
> Here the thing is when i try to read the holding registers 0-30 i am successfully reading but more than that i am unable to do.

Can you run with debug output on: ( e.g. `export DEBUG=modbus-serial` on linux ) and look at the messages ?

raviteja2503, 2019-03-27:
Hi I tried that `export DEBUG=modbus-serial` and i got the following messages.
when i try to read the register 3203.
[image]
when i tried to read the registers between 1-30 i got the following message.
[image]
What might be the reason for the issue.

yaacov, 2019-03-27:
The request looks OK, and the responce is "ILLEGAL DATA ADDRESS"
Are you sure address is currect ?

Request:

Part of Data Package | Description | Value
-- | -- | --
01 | Slave address | 0x01 (1)
03 | Function code | 0x03 (3) - Read Holding Registers
0C 83 | Starting address | 0x0C84 (3204)
00 01 | Quantity | 0x0001 (1)
76 B2 | CRC | 0x76B2 (30386)

Responce:

Part of Data Package | Description | Value
-- | -- | --
01 | Slave address | 0x01 (1)
83 | Error code | 0x80 + 0x03 (3) - Read Holding Registers
02 | Exception code | [02] ILLEGAL DATA ADDRESS
C0 F1 | CRC | 0xC0F1 (49393)

yaacov, 2019-03-27:
Ahhh, look at the address it's 3204 ....

raviteja2503, 2019-03-27:
Yes, The address is 3203 when i am trying with Modscan software i am getting the actual (Live) value on that register i.e the value which energy meter is showing in the screen.

sigma-design, 2019-03-27:
Could it be never ending Modbus confusion between 'address' and 'register' number, i.e. offset 0 or 1 ?
Typically, notation 43203 means "Holding register" number 3203, but actual address in the request/response would be 3202. 
So, if the function readHoldingRegisters requires 'address' as an argument, then it should be 3202.

raviteja2503, 2019-03-27:
Thanks for the quick reply.
I tried with that also but i am facing same issue. the issue is not only with that particular address, when i try to read the address more 30 i am getting the issue.

yaacov, 2019-03-27:
Can you look at  the raw message  Modscan software send and receive ... maybe we can understand something, the request for 3202~3 looks right to me :-( and should work ...

raviteja2503, 2019-03-27:
Now i am getting the output, thanks for the help.
The issue is when i am giving length 6 i am able to get the value and when i want to read that singe register i am facing the issue.
The code snippet which is working is.
```
client.readHoldingRegisters(3202, 6)
            .then(function (d) {
                console.log("Receive:", d.data);
            })
            .catch(function (e) {
                checkError(e);
                console.log(e.message);
            });;
```
May i know why this problem occurred, Here i am getting values in decimal format i want in floating format how this can achieved.

yaacov, 2019-03-27:
> May i know why this problem occurred

:question:  we will have to look into it, I think that looking at raw data from  Modscan may help to understand ...

raviteja2503, 2019-03-27:
> ❓ we will have to look into it, I think that looking at raw data from Modscan may help to understand ...
Ok, I will look into the raw data from Modscan part.
Thanks for the help

sigma-design, 2019-03-27:
If you are trying to read 32-bit floating point value, then it occupies 2 Modbus registers, so requesting odd number of registers is supposed to be rejected by the slave unit as an invalid request (your initial code shows request just for 1 register).
To read 32-bit values the request must be for even number of registers.

raviteja2503, 2019-03-29:
OK, Now i got it and i am able to read data successfully. Thanks for that help.

closed 2019-03-29 by raviteja2503