yaacov/node-modbus-serial #262: Max register quantity

Woiife, 2019-04-02:
Hello,

for my own implementation i would like to know how this project handles the maximum register quantity for each function code.

E. g: 2000 at once for Read Coils (FC 0x01). I have tried to review the code but could not find anything (Asking specially for TCP client).

I dont have to much exp with modbus, so I am not sure if it plays no role for Modbus TCP, or if it is critical at all.

If i want to read as many registers at once as possible, do I have to take care for myself to not requesting to many registers at once? What about the minimum and maximum register adresses?

Thank you in advance

Kind regards

yaacov, 2019-04-02:
AFAIK, total number of colis is limited by 0 - 0xffff, same for register address, but you should look at the machine specs/docs, some machines have no or very few valid addresses.

Woiife, 2019-04-02:
Wow what a quick response! Thank you very much.

Ofc I will take the individual device into account. But my question is really about the maximum quantity. If you look in the modbus specs 

http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

You see on page 5 that the maximum PDU size is 253. If you compare with the FC03 they limit the maximum quantity of registers to 125. Since in the response, the 2*125 registers + the 3 byte header you end up with 253.

At this point im am not sure how different devices will behave, if you would request more than 125 registers at once. Since the response would be bigger than the 253.
I saw a implementations for an other protocol with simular restrictions. There the module seperates the the request in multible requests (e. q If you request 130 register it would fire a 125 and a 5 register request). In the application layer you dont notice this seperation.

Is this also the case here? Or does the module fire the 130 register request as it is and I get or dont get a expecion back from the device?

yaacov, 2019-04-02:
> You see on page 5 that the maximum PDU size is 253

Sound correct to me. so ... using 8 coils (one bit per coil) per byte :  8 * 253 = 2024,
You should test with > 2024 coils ... , you will need a device that support it.
  
> Is this also the case here?

No, :-) what you see is what you get, e.g. one request with length=N.

Woiife, 2019-04-02:
Thank you very much! It is ok as it is but you have to be aware.

closed 2019-04-03 by Woiife