yaacov/node-modbus-serial #263: Data reply checks also checks type

attakorns, 2019-04-15:
Hi there,

Thank you for a wonderful module! I'm using this inconjunction with firebase functions so that I can go to a particular URL on any browser and get values from my serial device. It is working great then I decided to modify the function so that it would take in request arguments from the URL such as the IP address, port, serial ID, register number, etc and use those to connect to a specific device and read a specific value.

After doing that I found that the function would return a 'Unexpected data error' error even though the 'expected' and 'got' values logged are the same. I then discovered that if I called the setID function and passed a number instead of the serial ID argument from the HTTP call it would work. I suspected that this is probably a type difference issue. So I searched the code and found that in index.js, lines such as:
if (address !== transaction.nextAddress || code !== transaction.nextCode) 
if (data.length !== transaction.nextLength) 
are all value and type checks as suspected and this failed because the arguments from the HTTP call are strings instead of integers.

I understand that these checks are strictly correct since we should always enforce the correct type being used and I can simply convert the HTTP arguments from string to integer but on the other hand, these checks, by logic, are meant to check only for differing values, so another part of me feels that these checks should just be != instead of !==.

yaacov, 2019-04-15:
Hi, Thanks for the comment :+1:  :heart: 

> so another part of me feels that these checks should just be != instead of !==.

Ahh... I do not want `!=` and `==` in the code, it will require a comment explaining why this should not be replaced coincidently by `!==` and `===` and a linter exception, not realy want to do it :-(

If you make a nice PR that address linter and comments and looks nice overall, I may re-consider :-) , but it will have to look really nice !

yaacov, 2019-06-19:
@attakorns hi, closing this one, feel free to make a PR that fix this issue if you feel like it :heart: , it will be merged if it will be nice :-)

closed 2019-06-19 by yaacov