yaacov/node-modbus-serial #264: All requests time out after combined packet

adam-nielsen, 2019-04-16:
In my code, I have multiple calls to `readHoldingRegisters`.  The first ones work fine, but later ones often time out.  If I catch the time out exception and repeat the call, I can call it 30+ times and every single call will time out.  Aborting the program and running it again immediately succeeds on the early calls (so the device is fine) but again usually times out on the later calls.

I have done some investigation with Wireshark, and the problem seems to be triggered when multiple requests are combined into the same outgoing TCP packet.  My Wireshark logs look like this:

* Request 1
* Response 1
* Request 2
* Response 2
* Request 3+4+5
* Response 3
* Response 4+5
* Request 6
* Response 6 - never received, timeout exception
* Request 7 - my retry
* Response 7 - never received, timeout exception

This repeats forever - no further Modbus packets are received by the code until it is terminated and restarted (although Wireshark shows the device responding correctly at all times).

It looks like Modbus requests sent in quick succession are being batched together into the same TCP packet, however when the responses come back they aren't correctly matched up and all further incoming packets are ignored.

Any ideas why this might be happening?

yaacov, 2019-04-16:
Hi, thanks for the issue :heart: 

I assume you use `connectTCP` , we made `connectTcpRTUBuffered` to help with cases that regular tcp does not work.

a - you can try using it, it may solve your case :tada: .
b - if it does not solve your case :cry: , you can take it as a starting point to solve you case and create a connection that do solve your case, once it works, send a PR with the fix :smiley: 

https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcprtubufferedport.js

Thanks

yaacov, 2019-04-16:
p.s.
When testing you can use debug mode:
`export DEBUG=modbus-serial`

https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

Jarwain, 2019-04-19:
How quickly are you sending requests? Like how many ms between each call to readHoldingRegisters?

I'm under the impression that modbus devices get buggy when polling more frequently than 100ms

adam-nielsen, 2019-04-30:
Thanks for the info!  I am already using `connectTcpRTUBuffered` so hopefully that's all good there.

The requests are being sent very quickly, but the Wireshark logs show the Modbus device is responding correctly, however the data it sends back never gets processed by `modbus-serial`.

I will see if I can investigate further and work out exactly where the problem is.  I think it might just be that `modbus-serial` cannot handle multiple Modbus replies in the same TCP packet.

yaacov, 2019-04-30:
Thanks for the update,

If you are not using it already :-) , try using the DEBUG option https://github.com/yaacov/node-modbus-serial/issues/264#issuecomment-483514174

yaacov, 2019-10-12:
closing as stale

closed 2019-10-12 by yaacov