yaacov/node-modbus-serial #265: use of await in modbus-serial npm

SoulKa, 2019-04-26:
I just installed the node package on my raspberry pie that uses node v6. Unfortunately "await" is not supported by node v6 which causes your module to crash.

The "await" in the node package is used in server/servertcp_handler.js line 781:
var userObjects = await new Promise((resolve, reject) => {

Your npm page says:
This module has not been tested on every single version of NodeJS. For best results you should stick to LTS versions, which are denoted by even major version numbers e.g. 4.x, 6.x, 8.x.

Are the versions < 7.6 no longer supported?

I think a new babel compile with the master of your git would be sufficent to make the node package work again for these versions.

Greetings, Louis

yaacov, 2019-04-26:
Hi, thanks for the issue :+1: , 

I think that the offending method can be re factored to not use await, we will need to look into that :-) @debevv  can you take a look ?

yaacov, 2019-10-12:
Will be happy to get a pull request fixing that ... closing for now.

closed 2019-10-12 by yaacov