yaacov/node-modbus-serial #266: CRC Error in Telnet Client readCoils

ol-bones, 2019-05-02:
Using readCoils I get CRC error from index.js:274.

The data is returned as expected; commenting lines 271 - 276 removes the error and the data can be utilised.

Data: <Buffer cc 00 01 02 ff 00 c5 cc>

yaacov, 2019-05-02:
Hi, thank you for the issue :heart: 

The data you provided here has a CRC error, C5 CC should be F0 1B.

Part of Data Package | Description | Value
-- | -- | --
CC | Slave address | 0xCC (204)
00 | Function code | 0x00 (0)
01 02 FF 00 | Data |  
F0 1B | CRC | 0xF01B (61467)

ol-bones, 2019-05-02:
Would you say this may indicate a client or server error? Other calls don't raise this

yaacov, 2019-05-02:
hi,

> Would you say this may indicate a client or server error? Other calls don't raise this

You should look at more request <-> response sets, it's very hard to tell from one call ...

ol-bones, 2019-05-08:
Update for anyone who comes across this,

The CRC code is fine, my server seems to mix up function codes for some reason e.g.

Request: 00 01 01 92 00 01 [5c 0a]
Response: 00 01 02 ff [00 c5]

This CRC is incorrect, but would be correct for 00 04 02 ff.

closed 2019-05-08 by ol-bones

yaacov, 2019-05-08:
Thanks for the update :+1: