yaacov/node-modbus-serial #269: ModbusRTU.ServerTCP on connection event

onur2677, 2019-05-20:
i can't see on wiki , is there a config for "connected" event  ? 🙏

yaacov, 2019-05-20:
Hi, thanks for the issue :+1: 

Not sure what the question is ?
Can you give an example.

onur2677, 2019-05-20:
Hi !

for example;

```js
var serverTCP = new ModbusRTU.ServerTCP(options)

serverTCP.on('connect', function (err) {

})

```

yaacov, 2019-05-20:
IIUC no, see:
https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp.js#L218

When `ServerTCP(options)` have a connection, we immediately start listening for "data".

yaacov, 2019-05-20:
Also see:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js

onur2677, 2019-05-20:
hmm ok i got it. thanks for answering my question.  :pray:

yaacov, 2019-05-20:
np, have fun :white_flower:

closed 2019-05-20 by onur2677