yaacov/node-modbus-serial #271: ENETUNREACH when running local logger

alexjoyner, 2019-05-30:
I am working on a project that is required to be completely local.  For a few days I ran the logger example program while my server had connection to the internet, and everything worked without any failures.  I checked the logs of my program and several times the logger hit time out errors, but the reconnect function worked perfectly to get the logger back online.

Then I ran a full test and completely disconnect the internet connection from the logger.  Now, the logger completely fails every couple hours with the following output.

```
we have to reconnect
Timed Out
connect ENETUNREACH 192.168.2.100:502
```

The logger has a connection to the that address which is the Modbus PLC I am reading from.  Although the server runs the reconnect function and it seems to have proper timeouts in place, something is still causing the program to hang with this error.

To make this part as simple as possible for now, the logger program is identical to [the modbus example](https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js)

yaacov, 2019-05-31:
Hi,
Thanks for the issue :white_flower: 

did you try to add ENETUNREACH in line https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js#L10 did it made a difference ?

alexjoyner, 2019-05-31:
Yes I did, and oddly there wasn't any effect. Also, Even before I added the error to the array, the program still ran the we need to reconnect log.

yaacov, 2019-05-31:
I think the next steps to clear up what happen here, can be:

a - Running with debug on ( `export DEBUG=modbus-serial` )
b - Adding more `console.log`'s to find exactly where the "net unreachable" error happen, it may not happen where we think :-)
c - Try different nodejs versions ( e.g. different implementation of `net.connect` )

Having more data may help narrow up the potential week points ...

alexjoyner, 2019-06-04:
@yaacov So adding ENETUNREACH to the array actually fixed the error.  The issue wasn't actually an issue with the program, it was a personal issue with docker.  With that said, is there a reason that current error list doesn't include NET errors like that.  If there isn't, I'd like to make a PR to update the list to:

```
const networkErrors = [
  'ESOCKETTIMEDOUT',
  'ETIMEDOUT',
  'ECONNRESET',
  'ECONNREFUSED',
  'EHOSTUNREACH',
  'ENETRESET',
  'ECONNABORTED',
  'ENETUNREACH',
  'ENOTCONN',
  'ESHUTDOWN',
  'EHOSTDOWN',
  'ENETDOWN',
  'EWOULDBLOCK',
  'EAGAIN',
];
```

yaacov, 2019-06-04:
> is there a reason that current error list doesn't include NET errors like that. 

Not that I can remember :-)

>  I'd like to make a PR

Great :cake:

yaacov, 2019-06-04:
Fixed by #272

closed 2019-06-04 by yaacov

alexjoyner, 2019-06-04:
Awesome first experience with PR's! Glad I could help.