yaacov/node-modbus-serial #276: Writecoil timeaout

jakubpycha, 2019-07-19:
Hello,
i have problem with writeCoil method timeout. 

I have a simple start function and use promise api through RTU buffered :
```
export function start() {
    log.debug("Modbus polling started");
    const port = process.env.MODBUS_PORT || "COM8";
    client.connectRTUBuffered(port, { baudRate: 19200 }, async ()=> {
      log.info("Modbus connected to " + port);
      try {
          client.setID(85);
          await client.writeCoil(101, true);
      } catch (e) {
        log.error(e);
      }
    });

}
```

client.writeCoil always throws TransactionTimedOutError, but response is by log received.

LOG:
```
[1563525728051] INFO  (28660 on DESKTOP-P78P5OJ): Modbus connected to COM8
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 55 05 00 65 ff 00 91 f1>,
  modbus-serial   unitid: 85,
  modbus-serial   functionCode: 5 } +0ms
[1563525728096] INFO  (28660 on DESKTOP-P78P5OJ): MQTT broker byl pripojen
  modbus-serial { action: 'receive serial rtu buffered port',
  modbus-serial   data: <Buffer 55 04 01 00 00 79>,
  modbus-serial   buffer: <Buffer 55 04 01 00 00 79> } +364ms
[1563525731057] ERROR (TransactionTimedOutError/28660 on DESKTOP-P78P5OJ):
    message: "Timed out"
    errno: "ETIMEDOUT"
```
Timeout value changing is without change.
Other read methods like client.readInputRegisters working well and correctly.

**modbus-serial: 7.6.0
node.js: v10.16.0**

yaacov, 2019-07-19:
Hi, thanks for the issue :+1: 

A. Are you sure the device actually answers currectly / no other devices write on the same line ?
The data in the debug printout  `55 04 01 00 00 79` is not a valid packet, their can be many reasons for that.
You can start by removing any other devices you have on line (if you have any :-)) and check if the problem is not related to other device connected to the same line.
 
B. Can you try the `Promise` convention of using `connectRTUBuffered` ?
For example `...connectRTUBuffered( ... ).then( ...),catch( ... )`

See:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger_complete.js

jakubpycha, 2019-07-19:
> Hi, thanks for the issue 👍
> 
> A. Are you sure the device actually answers currectly / no other devices write on the same line ?
> The data in the debug printout `55 04 01 00 00 79` is not a valid packet, their can be many reasons for that.
> You can start by removing any other devices you have on line (if you have any :-)) and check if the problem is not related to other device connected to the same line.
> 
> B. Can you try the `Promise` convention of using `connectRTUBuffered` ?
> For example `...connectRTUBuffered( ... ).then( ...),catch( ... )`
> 
> See:
> https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger_complete.js

Thanks for your quickly answer.

A: On the modbus bus is only one slave and one master device. I  will take a look to response packet, but in CAS modbus scanner this request response working correctly.

[image]

B: Ok, I will try it.

yaacov, 2019-07-19:
you can set the scanner to only do write coil, and then compare the in / out buffers in the two softwares

for example in the screen-shot I see address 0x64 (100) while in the in the example it's 0x65 (101)

jakubpycha, 2019-07-19:
Sorry, it was not the same request.

This should be same as request and response from code and in the CAS working correctly.

[image]

yaacov, 2019-07-19:
you should compare the requests and responses to see the different.
in the example you posted the response is:
55 **04** 01 00 00 79
while in the screenshot it's:
55 **05** 01 00 00 79

Can you check why this happen ?

yaacov, 2019-07-19:
a. try connectRTU without buffering.
b. try adding debug printouts in:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/rtubufferedport.js

jakubpycha, 2019-07-19:
a: If i try connectRTU error is:
```
[1563533624948] ERROR (24784 on DESKTOP-P78P5OJ): Data length error, expected 8 got 6
    Error: Data length error, expected 8 got 6
        at SerialPort.<anonymous> (C:\Users\jakub\miwa\bms-gateway\node_modules\modbus-serial\index.js:302:42)
        at SerialPort.emit (events.js:198:13)
        at addChunk (_stream_readable.js:288:12)
        at readableAddChunk (_stream_readable.js:269:11)
        at SerialPort.Readable.push (_stream_readable.js:224:10)
        at binding.read.then.bytesRead (C:\Users\jakub\miwa\bms-gateway\node_modules\@serialport\stream\stream.js:384:12)
```
I tried also promise connectRTUBuffered, but without difference.

I do not know how it is possible but now code returns the same result as CAS, but timeout is still here.

```
[1563533746322] INFO  (28212 on DESKTOP-P78P5OJ): Modbus connected to COM8
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 55 05 00 65 ff 00 91 f1>,
  modbus-serial   unitid: 85,
  modbus-serial   functionCode: 5 } +0ms
[1563533746353] INFO  (28212 on DESKTOP-P78P5OJ): MQTT broker byl pripojen
  modbus-serial { action: 'receive serial rtu buffered port',
  modbus-serial   data: <Buffer 55 05 01 00 00 79>,
  modbus-serial   buffer: <Buffer 55 05 01 00 00 79> } +411ms
[1563533749331] ERROR (TransactionTimedOutError/28212 on DESKTOP-P78P5OJ):
    message: "Timed out"
    errno: "ETIMEDOUT"
```

yaacov, 2019-07-19:
> but timeout is still here.

The correct response to `55 05 00 65 ff 00 91 f1` is an echo `55 05 00 65 ff 00 91 f1`
https://en.wikipedia.org/wiki/Modbus#Function_code_5_(force/write_single_coil)

Any other response is wrong, and will cause timeout, If you know your device, and know it does not answer correctly to FC05 but do change the coil status,  you can ignore the timeout, may be send a get coil status request to be sure.

I do not know why CAS accept wrong answer ... maybe they know it's common to answer incorrectly to FC05 and ignore it by default ?

closed 2019-10-12 by yaacov