yaacov/node-modbus-serial #278: Break on Node 12 and Linux Alpine 3.9

ruizmarc, 2019-08-13:
Compilation breaks with Node 12 on Linux Alpine 3.9.

It seems serialport bindings are crashing. I have checked serialport bindings new releases and it seems they have updated the library to support Node 12. Maybe the problem can be solved just by updating dependency versions. (https://github.com/serialport/node-serialport/blob/master/packages/bindings/CHANGELOG.md#bug-fixes-1)

Here you have the stack trace:

https://pastebin.com/y8Ntrp39

yaacov, 2019-08-13:
Thank you for the issue :heart: 

Can you make a pull request updating the dependencies ?

ruizmarc, 2019-08-13:
There are some breaking changes that I don't really know how they may affect your library. I can do the pull request, but I cannot assure it won't have side effects...

ruizmarc, 2019-08-13:
I will have a try now, just to check if it compiles with the new versions ;)

ruizmarc, 2019-08-13:
We were not lucky, I have just updated the dependencies and it is working properly on Node 12 and MacOS (with some warnings), but not working on Node 12 and Linux Alpine :( 

It seems serialport have problems with Node 12

yaacov, 2019-08-13:
do you need serial port ?
we have a version without it:
https://www.npmjs.com/package/modbus-serial/v/7.6.0-no-serial-port

ruizmarc, 2019-08-13:
Wow! I didn't see you had a version without it :D I'm only using it over TCP. You made my day!

closed 2019-10-12 by yaacov