yaacov/node-modbus-serial #279: Can not install modbus-serial

kktuo, 2019-09-21:
Hi all, 

   I am trying install modbus serial in mi Raspberry 3b+, but i always have same error

pi@rpi:~/$ npm install modbus-serial
(node:1672) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.19.66-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install" "modbus-serial"
npm ERR! cwd /home/pi/powerwall
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405

I check with --unsafe-perm --build-from-source and same problem

Can somebody help me?
Thanks

yaacov, 2019-09-21:
Hi, thanks for the issue,

can you print out the error when you try :
```
$ npm install modbus-serial@7.6.0-no-serial-port
```

yaacov, 2019-09-21:
also look here:
https://github.com/nebrius/raspi-serial/issues/8#issuecomment-419503710

you may need a newer version of npm:
```
sudo npm install -g npm
```
may help

kktuo, 2019-09-22:
Thank you friend, 

 With this commands same error, i just update writing it

sudo npm install -g n
sudo n estable

and i trying again

 npm install modbus-serial

> @serialport/bindings@2.0.8 install /home/pi/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: se entra en el directorio '/home/pi/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: se sale del directorio '/home/pi/node_modules/@serialport/bindings/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ modbus-serial@7.6.0
added 74 packages from 43 contributors and audited 405 packages in 83.599s
found 25 vulnerabilities (3 low, 14 moderate, 8 high)
  run `npm audit fix` to fix them, or `npm audit` for details

yaacov, 2019-09-22:
what errors do you get for https://github.com/yaacov/node-modbus-serial/issues/279#issuecomment-533815562 ?

kktuo, 2019-09-22:
Now seems problems with package.json

pi@rpi:~ $ npm install modbus-serial@7.6.0-no-serial-port
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.jso                             n'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ modbus-serial@7.6.0-no-serial-port
added 3 packages from 4 contributors and audited 3 packages in 10.218s
found 0 vulnerabilities

yaacov, 2019-09-22:
> Now seems problems with package.json

Do you have a `package.json` ? 
if you do not, the warnings look ok to me ... I think it's installed successfully this time ( without serial port support )

yaacov, 2019-10-29:
closing stale, feel free to reopen on new data.

closed 2019-10-29 by yaacov