yaacov/node-modbus-serial #280: error: 'error processing response'

Louro, 2019-09-24:
Hey i'm trying to make a simple modbus TCP server and atm i have 1 client pulling a coil at address 0, but i'm getting error: 'error processing response'.

Any idea what can be wrong?

Debug log:
 modbus-serial { action: 'socket data',
  modbus-serial   data: <Buffer 00 2c 00 00 00 06 01 01 00 00 00 01> } +99ms
  modbus-serial { action: 'receive',
  modbus-serial   data: <Buffer 01 01 00 00 00 01 fd ca>,
  modbus-serial   requestBufferLength: 8 } +1ms
  modbus-serial {"action":"receive","data":{"type":"Buffer","data":[1,1,0,0,0,1,253,202]}} +0ms
  modbus-serial request for function code 1 +1ms
  modbus-serial { error: 'error processing response',
  modbus-serial   unitID: 1,
  modbus-serial   functionCode: 129 } +0ms
  modbus-serial { action: 'server response',
  modbus-serial   unitID: 1,
  modbus-serial   functionCode: 129,
  modbus-serial   responseBuffer: '0181044193' } +0ms
  modbus-serial { action: 'send', data: <Buffer 01 81 04 41 93> } +0ms
  modbus-serial {"action":"send string","data":{"type":"Buffer","data":[1,129,4,65,147]}} +0ms
  modbus-serial { action: 'socket data',
  modbus-serial   data: <Buffer 00 2d 00 00 00 06 01 01 00 00 00 01> } +98ms
  modbus-serial { action: 'receive',
  modbus-serial   data: <Buffer 01 01 00 00 00 01 fd ca>,
  modbus-serial   requestBufferLength: 8 } +0ms
  modbus-serial {"action":"receive","data":{"type":"Buffer","data":[1,1,0,0,0,1,253,202]}} +1ms
  modbus-serial request for function code 1 +2ms
  modbus-serial { error: 'error processing response',
  modbus-serial   unitID: 1,
  modbus-serial   functionCode: 129 } +1ms
  modbus-serial { action: 'server response',
  modbus-serial   unitID: 1,
  modbus-serial   functionCode: 129,
  modbus-serial   responseBuffer: '0181044193' } +0ms
  modbus-serial { action: 'send', data: <Buffer 01 81 04 41 93> } +0ms
  modbus-serial {"action":"send string","data":{"type":"Buffer","data":[1,129,4,65,147]}} +1ms

yaacov, 2019-09-24:
hi, thanks for the issue

does the example server works for you ?
https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js

Louro, 2019-09-24:
Hi thank you for such a fast reply.

I can confirm it works as expected with the example server.js.

I will keep learning and testing more.

Thanks.

closed 2019-09-24 by Louro