yaacov/node-modbus-serial #281: How to read holding registers with the type String and addresses like 412350.20L

guilhermedecampo, 2019-09-25:
Hi @yaacov first of all thanks for the time to create this package and share.

**Overview**

I'm working with a schneider electric PLC M580 that has String type and this kind of address 412350.20L. I'm pretty noob about PLCs and how to convert these "complex" data types... to read "short" types work just fine the output of the lib.

Accordingly to the table below the L means LoHi order byte. But I have no idea how to convert to usable data yet.

[image]

**Specific case and the results:**

Address: 412380.10L
Stores: BCA111719
What should send me: 66 67 65 49 49 49 55 49 57

When I search for 12380 + 10 length on holdingRegisters (4 in the beginning of the address means holdingRegister) I receive from the lib is:

[image]

I know it's wrong the start... should be 12379 + 10, but we can reason about it.

**The questions:**

- Why the lib is converting to 5 digits? 
- How can I convert them so I can use `String.fromCharCode` to know which character is saved on the addresses?
- And why there are some addresses with the "correct" charCode

**Testing in my machine:**

There is an easier way to test these? I have a modbus server here working but doesn't have these complex types.

[image]

yaacov, 2019-09-25:
here you can see an example of parsing incoming data as floats:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js#L30

p.s.
you did not send the response buffer in your issue ... so it's hard to understand if what you got resembles what you expected. 
you can try `buf.toString('ascii')` and see what you get ?

yaacov, 2019-09-25:
dec (Int)     | hex             | dec (Hi,Lo) | ascii
---------|----------------|-----|-------------
12609 | 0x31, 0x41  | 49, 65 | 1, A
12593 | 0x31, 0x31  |  49, 49  | 1, 1
12599 | 0x31, 0x37  |  49, 55 |  1, 7
57 | 0x 39                 |  57        | 9

You got 'A111719', you are missing the `BC` at the beginning

guilhermedecampo, 2019-09-25:
Thank you @yaacov.

I'm gonna try to handle buffer! (I have access just some hours of the day)

Seems that in the case of the `float` type... I'll need to use `buffer.readFloatLE(20)` 412380.20L. Right?

Wow... this table! It's exactly this! Can you explain please why some have 2 and other 5? 🚀

guilhermedecampo, 2019-09-25:
Ah actually, we know that ASCII uses just HEX with 2 chars.

guilhermedecampo, 2019-09-25:
Example of the reducer to return the correct string!

```
[12609, 12593, 12599, 57].reduce((accInt, int) => accInt + Number(int).toString(16).match(/.{1,2}/g).reduce((accHex, hex) => accHex + String.fromCharCode(`0x${hex}`),''),'')
```

yaacov, 2019-09-25:
> Can you explain please why some have 2 and other 5? rocket

they all have same length

bytes | hex | dec
----|------|------
0000 0100 1101 0010 (16 bits, 2 bytes) |  [0]4d2 ( 3 digits ) | [00]1234 ( 4 digits )

the [000]57 you see in dec is actually 5 digits if you count the leading zeros :-)  

bytes | hex | dec
----|------|------
0000 0000 0011 1001 (16 bits, 2 bytes) |  [00]39 ( 2 digits ) | [000]57 ( 2 digits )

yaacov, 2019-09-25:
> Example of the reducer to return the correct string!

did you try taking the buffer and doing just one  `.toString('ascii')` on the complete buffer ?

guilhermedecampo, 2019-09-25:
Not yet. I'll have access to it in 1h or so. Let you know.

yaacov, 2019-09-25:
In case using `buffer` will get you a mixed-up string, you can try this:
 
``` es6
> buffer = [12609,12593,12599,57].reduce((buf, val, index) => { buf.writeUInt16LE(val, index * 2); return buf }, Buffer(20))
<Buffer 41 31 31 31 37 31 39 00 00 00 00 00 00 00 00 00 00 00 00 00>

> console.log(buffer.toString('ascii'))
A111719
```

guilhermedecampo, 2019-09-25:
Hey @yaacov we could read via `buffer.toString('ascii')`!

The only issue is because it's little-endian. So the chunks of 2 chars are in the wrong order. 

```
data.buffer.toString('ascii').match(/.{1,2}/g).reduce((acc, chunk) => acc + chunk.split('').reverse().join(''), '')
```

I see that you're creating a new buffer. Do you think is less hacky than the match/reduce?

yaacov, 2019-09-25:
> I see that you're creating a new buffer. Do you think is less hacky than the match/reduce?

No, I like it better because it's more readable to **me** you should do what is better for you :-)

guilhermedecampo, 2019-09-25:
Thank you, we keep it opened or close?

yaacov, 2019-09-25:
hey, try:
```
buffer.swap16().toString('ascii')
```

closed 2019-09-25 by yaacov

guilhermedecampo, 2019-09-25:
Gonna try tomorrow! 

Thanks for all the help, you rock. 🚀 

Can I pay you a beer or something? 🍻

yaacov, 2019-09-25:
> Can I pay you a beer or something? beers

Thanks :exclamation: 

Didn't set anything up (yet) , may be a good idea to set something for the next time,
I may call you on it, If I get to Brazil :-)

guilhermedecampo, 2019-09-26:
For sure! Let me know.

guilhermedecampo, 2019-09-26:
Yeah `swap16` worked @yaacov! Had to put some `trim` too to remove empty spaces
```
          res.buffer
            .swap16()
            .toString('ascii')
            .trim()
```