yaacov/node-modbus-serial #285: function getCoil cannot access outer scope variables

Louro, 2019-09-30:
Hey

When using getCoil function, i cannot access any of the outer scope variables, not event "this." is defined, while if i use setRegister works without a problem.

            getCoil: function(addr) {
              debugger
              return (addr % 2) === 0;
            },
            setRegister: function(addr, value) {
              this.onMessage('setRegister', addr, value);
              return;
            },

yaacov, 2019-09-30:
hi, thanks for the issue :heart: 

In `getCoil` [L91](https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L91) we assign the function into a variable.
while in `setRegister` we use the function as is [L501](https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L501)

You can try using `getCoil` directly instead of `vectorCB` in [L133-137](https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L133) and see if that change anything ?

Louro, 2019-09-30:
Hi, i think thats also the problem, somehow it loses the scope its in. 

If I clone the git, how can i recompile and overwrite my current installation of modbus-serial?

I'm using the modbus inside another problem and i'm very need to this nodejs.

thank for such a fast response once more.

yaacov, 2019-09-30:
depending of how your package depend on the modbus package ... , this options look like a start :

option a - go to your `node_modules` directory, find the `modbus-serial` code and edit it.

option b - after git clone, run `npm install` and then change the `requires` in your package to point to the local git code using `file` prefix:
```
"dependencies": {
  "package name": "file:../../oresoftware/your package",
},
```
option c ... ?

see:
https://medium.com/@the1mills/how-to-test-your-npm-module-without-publishing-it-every-5-minutes-1c4cb4b369be

Louro, 2019-09-30:
Hey

Thanks, that works great.

It was indeed the problem, by using the var to create the vectorCB.

Is there a simple way to fix this issue, without using the if else?

                if (vectorCB.length === 3) {
		  if(fc === 1)
		        vector.getCoil(address + i, unitID, cb);
		    else if (fc === 2)
		        vector.getDiscreteInput(address + i, unitID, cb);
                }
                else {
                    var promiseOrValue;
		    if(fc === 1)
		        promiseOrValue = vector.getCoil(address + i, unitID, cb);
		    else if (fc === 2)
		        promiseOrValue = vector.getDiscreteInput(address + i, unitID, cb);

                    _handlePromiseOrValue(promiseOrValue, cb);
                }

yaacov, 2019-09-30:
> that works great.

:+1:

> Is there a simple way to fix this issue, without using the if else?

I do not know :-) make a pull request, and try different things until you like it ... if we can't find a good way now, we can fix it later ...

yaacov, 2019-10-11:
close by #286

closed 2019-10-11 by yaacov