yaacov/node-modbus-serial #299: Unable to install the module

codyc1515, 2019-10-28:
I am unable to install the module. I'm running node v13.0.1 with npm v6.12.0.

The command I am using is `sudo npm install -g --unsafe-perm --build-from-source modbus-serial`.

Logs & errors: https://gist.github.com/codyc1515/cf80a4264142734a328947b429677632

yaacov, 2019-10-28:
a - try installing without serialport `sudo npm install -g modbus-serial@7.7.3-no-serial-port`
b - look for a way to install `serialport` on your system: https://serialport.io/docs/guide-platform-support

if you succeed in installing `serialport` try to understand why it can't install as a dependency of modbus-serial, we may be able to fix that.

yaacov, 2019-10-29:
@codyc1515 ?

yaacov, 2019-11-05:
closing, stale

closed 2019-11-05 by yaacov