yaacov/node-modbus-serial #300: Reading Single Holding Register Problem

Catutigernos, 2019-11-05:
Hi,
I tried to read one holding register.  

This code is not working. Throw an exception.
```
client.setID(1);
client.connectTCP("169.254.102.169", { port: 502 })
.then(()=>{
  console.log("connected")
  client.readHoldingRegisters(2001, 1)
  .then(function(data) {
      console.log(data);
  })
  .catch(err => console.error(err));
})
.catch(err => console.error(err));
```

> Error: Modbus exception 3
>     at TcpPort.<anonymous> (D:\ScadaProject\scada-server-light\node_modules\modbus-serial\index.js:288:42)
>     at TcpPort.emit (events.js:182:13)
>     at Socket.<anonymous> (D:\ScadaProject\scada-server-light\node_modules\modbus-serial\ports\tcpport.js:70:20)
>     at Socket.emit (events.js:182:13)
>     at addChunk (_stream_readable.js:283:12)
>     at readableAddChunk (_stream_readable.js:264:11)
>     at Socket.Readable.push (_stream_readable.js:219:10)
>     at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)

but when I make read register length 2, code is working and value is correct. 

```
client.setID(1);
client.connectTCP("169.254.102.169", { port: 502 })
.then(()=>{
  console.log("connected")
  client.readHoldingRegisters(2001, 2)
  .then(function(data) {
      console.log(data);
  })
  .catch(err => console.error(err));
})
.catch(err => console.error(err));
```

> { data: [ 5793, 0 ], buffer: <Buffer 16 a1 00 00> }

yaacov, 2019-11-05:
Hi, thank you for the issue.

Modbus exception 3 means "A value contained in the query data field is not an allowable value for the slave."

Can you check the device documentation to see if it allows querying for length 1 on this address ?

Catutigernos, 2019-11-05:
Thanks for your response. I checked, but there is no such an explanation about it. And I tried code is below and it is working. I just change register from 2001 to 0. When register is 0,  working, but when register is 2001, throws exception
```

client.setID(1);
client.connectTCP("169.254.102.169", { port: 502 })
.then(()=>{
  console.log("connected")
  client.readHoldingRegisters(0, 1)
  .then(function(data) {
      console.log(data);
  })
  .catch(err => console.error(err));
})
.catch(err => console.error(err));
```

> { data: [ 0 ], buffer: <Buffer 00 00> }

yaacov, 2019-11-05:
> When register is 0, working, but when register is 2001, throws exception

Sounds like the device does not accept length 1 for address 2001 :-(
To check that, you can use another modbus client and see the response, or query for more documentation about the device.

yaacov, 2019-11-05:
you can also comment here with device info and register description, maybe someone reading here will know something about your device.

yaacov, 2019-11-10:
@Catutigernos hi, any news ?

yaacov, 2019-11-12:
@Catutigernos closing, feel free to re-open, or just comment here, if you have new info.

closed 2019-11-12 by yaacov

Catutigernos, 2019-11-14:
Hi Yaacov,
Sorry I have to do some works, so I can't look the project. I request 2 register and get first register, I only solve the problem with this solution. So, there is no any news.