yaacov/node-modbus-serial #303: Need to reading different data type

hunter7793, 2019-11-15:
Hello @yaacov. I'm using `modbus-serial` and it's working well, but when i'm reading data(for example) - i'm receiving only unsigned data type. Can you help me and tell how can i read signed or float data (for example Integer data type which have range -32767...32768).
Thank you

yaacov, 2019-11-15:
Hi, thanks for the issue :+1:

we have an example here:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js

you can get the data buffer [1] of the answer by accessing the `buffer`, then you can parse
the raw data using your home-brew algorithm, or usethe `buffer` module [2] for example:
``` js
 client.readInputRegisters(4, 12)
        .then(function(d) {
            var floatA = d.buffer.readFloatBE(0);
```

[1] https://store.chipkin.com/articles/how-real-floating-point-and-32-bit-data-is-encoded-in-modbus-rtu-messages
[2] https://nodejs.org/api/buffer.html#buffer_buf_readfloatbe_offset

hunter7793, 2019-11-15:
Thank you very much! i will try it and write you later

hunter7793, 2019-11-15:
i`ve succesfully reading right data type from the buffer . thank you so much for solution!
But i have one more question. Where should i write my buffer array which contains float or integer data type?

yaacov, 2019-11-15:
> i`ve succesfully reading right data type from the buffer

:+1: 

> Where should i write my buffer

can you give a code example ?

hunter7793, 2019-11-15:
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

var startAddress = 0,
      arrayOfMyData = [1], // not a buffer, just array of number types
      buf = Buffer.allocUnsafe(2),
      myBuffer = buf.writeInt16BE(arrayOfMyData[0],0) // i want to write this array

client.writeRegisters(startAddress, arrayOfMyData ,function(err) {
            if (!err) {
                console.log('Somethng wrong'
            } else {
                console.log('Done with writing');
            }

where or how should i write "myBuffer" tag?

yaacov, 2019-11-15:
> where or how should i write "myBuffer" tag?

the `writeRegisters` method also accept a buffer instead of an array, so you can do:
```
var startAddress = 0,
      arrayOfMyData = [1], // not a buffer, just array of number types
      buf = Buffer.allocUnsafe(2),
      myBuffer = buf.writeInt16BE(arrayOfMyData[0],0) // i want to write this array

client.writeRegisters(startAddress, myBuffer ,function(err) {
```

By the way, does your device accept different type of numbers ?

hunter7793, 2019-11-15:
> does your device accept different type of numbers ?

Yes, my device is accept different type. 
I`m trying to read holding registers and then change in received array some data and send it to holding registers area.

I'm tryied to send myBuffer to the holding registers area and it's works. 
Thank you very much for help

closed 2019-11-15 by hunter7793

yaacov, 2019-11-15:
Nice :+1: thanks for the feedback.

p.s.
If you have a small peace of code that you can add as an example, feel free to send a pull request adding it to the example directory [1] it may help the next person looking to send floats to a device.

https://github.com/yaacov/node-modbus-serial/tree/master/examples

wmertens, 2021-11-08:
I'm confused about addresses and lengths.

I'm trying to REALs and the doc says that I have them on position 40002, 40003 etc, all having length 1. Does that mean that these are 2-byte floats? I tried using readFloatBE but the values don't make sense.

yaacov, 2021-11-08:
you can start here:
https://simplymodbus.ca/FAQ.htm#Types
try to re-read docs of the device you are using for more information.

wmertens, 2021-11-08:
@yaacov ok thank you, I figured it out. I found `00 c8` is 180 which is 10x the desired value 18.0. So the manufacturer is storing numbers with 1 decimal precision by multiplying by 10 before storing and dividing when retrieving.