yaacov/node-modbus-serial #305: How i read multi idClient for same ip?

mariodiciccio, 2019-11-19:
Hi,
like from title:
clientDrago.connectTCP( ip, { port: port });
clientDrago.setID(1);
clientDrago.setID(2);
clientDrago.setID(3);
.....
clientDrago.readHoldingRegisters(0, 4);

or

clientDrago1 = new ModbusRTU();
clientDrago1.connectTCP( ip, { port: port });
clientDrago1.setID(1);
clientDrago1.readHoldingRegisters(0, 4);

clientDrago2 = new ModbusRTU();
clientDrago2.connectTCP( ip, { port: port });
clientDrago2.setID(2);
clientDrago2.readHoldingRegisters(0, 4);

....

same ip.

yaacov, 2019-11-20:
Hi, thanks for the review :heart:
Do you setID only after the read is resolved ?

mariodiciccio, 2019-11-20:
yes

clientDrago.connectTCP( ip, { port: port });
clientDrago.setID(1);
clientDrago.readHoldingRegisters(0, 4).then(function(d) {
            /* content */
        clientDrago.setID(2);
        clientDrago.readHoldingRegisters(0, 4).then(function(d) {
            /* content */
            clientDrago.setID(3);
            etc....
       });
 });

yaacov, 2019-11-20:
maybe the errors are not connected to switching id's ?

you can try running with error logging, see example here:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger_complete.js

This may give you more info that may help understand the problem.

mariodiciccio, 2019-11-20:
thanks, i try.

mariodiciccio, 2019-11-21:
Hi,
I tried, but the "script" stop working after 58 minutes without error message.
I have used:
- nwjs 0.42.1;
- nodejs 13.0.1
- chromium 78.0.3904.70

the script.js:
var display = document.getElementById('app');
var ModbusRTU = require("modbus-serial");

var ip = '192.168.100.1';
var port = 502;

clientDrago = new ModbusRTU();
clientDrago.setTimeout(3000);
clientDrago.connectTCP( ip, { port: port })
.then(() => {
 // empty
})
.then(function() {
  console.log("CONNECTED DRAGO");
  drago1();
})
.catch(function(e) {
  console.log("ERROR CONNECT DRAGO " + e.message);
});

function drago1() {
  display.innerHTML = new Date();
	
  clientDrago.setID(1);
  clientDrago.readHoldingRegisters(0, 4)
  .then(function(d) {
      let a = d.data;
      console.log("Receive Coils1:", a); // decimale
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago1',e.message);
  })
  .then(() => {
    drago2();
  });
}

function drago2() {
  clientDrago.setID(2);
  clientDrago.readHoldingRegisters(0, 4)
  .then(function(d) {
      let a = d.data;
      console.log("Receive Coils2:", a); // decimale
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago2',e.message);
  })
  .then(() => {
    drago3();
  });
}

function drago3() {
  clientDrago.setID(3);
  clientDrago.readHoldingRegisters(0, 4)
  .then(function(d) {
      let a = d.data;
      console.log("Receive Coils3:", a); // decimale
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago3',e.message);
  })
  .then(() => {
    drago4();
  });
}

function drago4() {
  clientDrago.setID(4);
  clientDrago.readHoldingRegisters(0, 4)
  .then(function(d) {
      let a = d.data;
      console.log("Receive Coils4:", a); // decimale
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago4',e.message);
  })
  .then(() => {
    drago5();
  });
}

function drago5() {
  clientDrago.setID(5);
  clientDrago.readHoldingRegisters(0, 4)
  .then(function(d) {
      let a = d.data;
      console.log("Receive Coils5:", a); // decimale
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago5',e.message);
  })
  .then(() => {
    drago6();
  });
}

function drago6() {
  clientDrago.setID(6);
  clientDrago.writeCoils(0, [0,0,0,0])
  .then(function(d) {
      console.log('write colis 6');
  })
  .catch(function(e) {
    console.error('Error Holding Registers Drago6',e.message);
  })
  .then(() => {
    drago1();
  });
}
 

thanks

yaacov, 2019-11-21:
are you running if from a browser ?
try running it using nodejs, maybe you will get more debug info:

for example, on linux it will look like - 
``` bash
$ export DEBUG=modbus-serial
$ nodejs your-script.js
```

mariodiciccio, 2019-12-10:
now its ok, i introduced a bug.
thank you very much

yaacov, 2019-12-10:
nice !  can we close this ?

closed 2019-12-23 by yaacov