yaacov/node-modbus-serial #306: Add Modbus UDP

Ekristoffe, 2019-11-21:
Hello, in some case we need to use Modbus over UDP connection (it work exactly like the Modbus TCP but using UDP.
Is it possible to also have a Modbus UDP function ?
Thanks

yaacov, 2019-11-21:
Sure :+1:

Look at the ports dir - https://github.com/yaacov/node-modbus-serial/tree/master/ports

You can take the tcpport [ https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js ] and change it until it works for you.

If you succeed please open a pull request adding the port.

Ekristoffe, 2019-11-21:
indeed i think i need to make a mix between tcpport.js and c701port.js (which use UDP) also why not making a buffered version ...

sirockin, 2020-02-08:
@Ekristoffe did you get anywhere with this? I'd be happy to collaborate.

closed 2020-02-10 by yaacov

Ekristoffe, 2020-05-05:
Hello all,
Sorry for the long idle time.
I have started to work on it.
I will see what is needed.
BTW what function is needed ?

commit 6bc71d3b (2020-02-10): implemented connectUDP() to fix #306 (#320)