yaacov/node-modbus-serial #310: Build error with NodeJS 13

MrJake222, 2019-12-09:
Hi, I have a problem installing your module with Node v13.3.0 and NPM 6.12.1.

This is the console output: https://gist.github.com/MrJake222/ea1896f781da054beaecfc38ec6e9987
And the mentioned file: https://gist.github.com/MrJake222/c31c1ce93439ceff83a3bdfa8b80c93c

Is this a problem with my Node being too new or what?
Anyway, any help appreciated.

MrJake222, 2019-12-09:
Tried with Node 12 and it works

yaacov, 2019-12-10:
Hi, thanks you for the issue :+1: 

Looks like serial-port version miss match :-(
a - you can make a pull request updating serial-port version to work with v13
b - if you do not use serial-port, you can remove the dependence on serial-port from the package file:
https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

closed 2020-01-02 by yaacov