yaacov/node-modbus-serial #311: как опрашивать по tcp  несколько slave устройств функцией?

bechterev, 2019-12-12:
как опрашивать по tcp  несколько slave устройств функцией 3, т.е. асинхронно, не получается больше 10 устройств за 1 секунду по 40 регистров получить. Выходит ошибка порт закрыт. Тоже самое на с++ для этих же устройств 100 миллисекунд занимет. Подскажите мне, что оптимизировать.

yaacov, 2019-12-12:
English :see_no_evil: :hear_no_evil: :speak_no_evil: 

a - Can you attach the code ?
b - Can you run with DEBUG on, to see the communication packets ?
c - Can you run with fireshark and see the delays ?

bechterev, 2019-12-18:
Hello, sorry for the fact that the question was not in English.
var ModbusRTU = require("modbus-serial");
var registerInit = require("./need_register");
Models.Devices.findAll({ where: { id: { [Sequelize.Op.gt]: '10' } } }).then(res => {
  res.map((value) => {
    //let clients = new ModbusRTU();
    clients.setTimeout(3000);
    clients.connectTCP(value.ip);
    job.push(clients, value['id'])
  });
  return res;
}).then(res => {

  var resultat = Object.assign(res);
  const getRegisters = async (resul) => {
    try {
      console.log(`start write into bd ${new Date()}`);
      if (flagbuff == false) {
        buff = await getBuffTable().then(resa => {
          let g = JSON.stringify(resa);
          let a = g.slice(1, g.indexOf(']') + 1);
          let b = JSON.parse(a);
          flagbuff = true;
          return b;
        });
        console.log('virt table');
      }
      let temp = Array();
      for (let device of resul) {
        let test;
        for (let i = 0; i < job.length; i++) {
          if (device.id == job[i]) {
            test = job[i + 1];
          }
        }     
        let dat = await getRegister(device, test);
        dat.unshift(device.id, device.name);
        temp.push(dat);
      }
.....

  const getRegister = async (dev/*, reg*/, client) => {
    try {
      client.setID(parseInt(/*1*/dev.name == 15 ? 1 : dev.name));
      let val =await client.readHoldingRegisters(260, 40/*reg, 1*/).then(async res => {
        return res.data;
      }).catch(e => {
        console.log(e.message);/*client.connectTCP("192.168.144.246", { port: 502 });*/
      })
      return val;
    }
    catch (error) {
      console.log(error, 'ошибка тут');
      return -1;
    }
  }

bechterev, 2019-12-18:
how to do several different clients with different ip addresses. What am I doing wrong?

yaacov, 2019-12-18:
The code you attached is complex, the two big problems are - 
a - you should use `await (try ... catch)` **or** `.then / .catch` **not together**.
b - `client.connect ... ` return a `Promise` you need to either `await` for it **or** handle it with `.then / .catch`

Try writing simpler test programs with only 3 clients, that start from very simple program and go to more complex slowly:

1. start with connecting to one device (wait for it to connect) and then reading it , then connecting to the next device ....

2. try to connect to all three devices (wait for all connection to finish) and then read all the clients.

3. write what worked for you in stage 2 in a loop (`map`, `foreach`, `for` ... what's best for you ).
then try to write the more complex program ( but this time with only `await` OR `.then` not both )

bechterev, 2019-12-19:
Hello, thank you Yaakov. I reworked the code, removed some of the promises and instead made asynchronous functions with await. As a result, I was able to get an acceptable result. 30 units, 3 of which are physical glands and 27 on the emulator Modbus 40 registers. Tell me have you had experience with 100 or more devices?

yaacov, 2019-12-19:
> Tell me have you had experience with 100 or more devices?

No :snowman: , if you do succeed in pooling efficiently large number of devices, please comment here with your notes, I'm sure other people trying to do this will be happy to have some reference.

yaacov, 2019-12-19:
p.s. if you use some sniffing tool ( like https://www.wireshark.org/) you can see where time is waisted, and try to profile your code ...

closed 2019-12-31 by yaacov