yaacov/node-modbus-serial #312: Question about modbus server

silvermoonswk, 2019-12-17:
Hi there

I am a newbie on nodejs and needs help on this modbus-serial module.

I can run the below successfully but I am struggling in working on the modbus server.

`// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
function run() {
    client.setID(1);

    client.readHoldingRegisters(22, 4, function(err, data) {
        console.log(data.buffer.readFloatLE().toString());
    });
}

// open connection to a serial port
client.connectRTUBuffered("/dev/tty.usbserial-14310", { baudRate: 4800 }, run);
`

What I want to achieve is:

1) Set up a server which has the modbus device connected
2) Use another pc to read the modbus data via the Internet (with the IP address)

I followed this
`// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a tcp line
client.connectTCP("my server ip", run);

    client.setID(1);
    client.readHoldingRegisters(22, 4, function(err, data) {
        console.log(data.buffer.readFloatLE().toString());
}`

But it can't get the correct value, and I changed the setID (1) -> e.g. (2,3,4...) also having the same result. And thus, I think the client did connect to the server but it failed to find the correct ("/dev/ttyusb")? So I would to ask for assistance for this problem. 

Cheers,
Edward

yaacov, 2019-12-17:
Hi, thank you for the issue :heart:

>    Set up a server which has the modbus device connected

see example here:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js

>    Use another pc to read the modbus data via the Internet (with the IP address)

The code example you wrote should work ...
a - you can use the debug output, try running with `export DEBUG=modbus-serial` 
see: https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js
b - you can try to use the `connectTcpRTUBuffered`
see: https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffertcp.js

silvermoonswk, 2019-12-17:
Hi yaacov

Thanks for the prompt reply!

Unfortunately I still couldn't get the correct result. My client can connect to the server, but the output isn't right... (shows 8022, 8023, 8024, 8025)... 

I should say, what I want to achieve is, use the client to connect to server's ip + server's specific usb port...So it's like client.connectTcp("IP address") & connectRTU("/dev/ttyUSB0").

Thanks for the help again!
Edward

yaacov, 2019-12-17:
> Unfortunately I still couldn't get the correct result.

Did you try running with debug on ? do you have some other way to sniff the communication (wireshark) ?
Do you have some other client that read data correctly from the device, are you sure the device doesn't actually have this data at this address ?

silvermoonswk, 2019-12-17:
Hi

1) I can't turn debug mode on, it said SyntaxError: Unexpected token export

2)If I run 

'// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
function run() {
    client.setID(1);

    client.readHoldingRegisters(22, 4, function(err, data) {
        console.log(data.buffer.readFloatLE().toString());
    });
}

// open connection to a serial port
client.connectRTUBuffered("/dev/tty.usbserial-14310", { baudRate: 4800 }, run);'

on the server side, it can output the correct result.

yaacov, 2019-12-17:
so you connect to the **same device** -
once using usb `Serial` connection, and once using TCP ?

silvermoonswk, 2019-12-17:
Correct

Server side PC:
1) use connectRTU('/dev/xxxx) -> works 
2) run as server; use client side PC to connect the server -> failed

yaacov, 2019-12-17:
> run as server

can you attach the server code ?

silvermoonswk, 2019-12-17:
Sure.

/* eslint-disable no-console, no-unused-vars, spaced-comment */

// create an empty modbus client
//var ModbusRTU = require("modbus-serial");
var ModbusRTU = require("modbus-serial");
var vector = {
    getHoldingRegister: function(addr) { return addr; },
};

// set the server to answer for modbus requests
console.log("ModbusTCP listening on modbus://0.0.0.0:8502");
var serverTCP = new ModbusRTU.ServerTCP(vector, { host: "0.0.0.0", port: 8502, debug: true, unitID: 1 });

serverTCP.on("initialized", function() {
    console.log("initialized");
});

serverTCP.on("socketError", function(err) {
    console.error(err);
    serverTCP.close(closed);
});

function closed() {
    console.log("server closed");
}

yaacov, 2019-12-17:
:smile: 
```
 getHoldingRegister: function(addr) { return addr; },
```
this function should actually do something ...
try returning 1234:
```
 getHoldingRegister: function(addr) { return 1234; },
```

you should get 1234 if you query the server ...

to do what you want you need to implement querying using usb, and returning the value ...

silvermoonswk, 2019-12-17:
Thanks. Yes I can get the value 1234 from the client side PC now.

`to do what you want you need to implement querying using usb, and returning the value ...`

May I ask do you have any example code? Thanks

yaacov, 2019-12-17:
> May I ask do you have any example code? Thanks

No, I think you will more have fun figure it out yourself :-)

Hint: the function need to do a request on a usb-serial client, and return the answer it get from the usb client. see the usb client code to do that ... you can also see an example of running more complex funtions here:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/async_server.js

silvermoonswk, 2019-12-17:
Got it! Many thanks for the help :)

closed 2019-12-17 by silvermoonswk