yaacov/node-modbus-serial #313: modbus : discover all the device Master and slave and generating a json file with all the devices and data model (parameters supporting)

yboujraf, 2019-12-23:
Dear @mwillbanks , @adam-nielsen , @ilang , @relovution , @brandly 

I am working on a proprietary protocol translator from /to modbus.

I need to discover all the modbus device master/slave and get the list of parameters they are able to support (finally the data model) for each devic discovered.

It's our data model for one parameter it could be string, boolean, real, integer, ....

```
{

                    "number": 0,
                    "identifier": "ParameTypeInteger",
                    "description": "",
                    "value": 0,
                    "minimum": -4096,
                    "maximum": 0,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 32,
                    "isOnline": true,
                    "formula": "1*\n1/",
                    "default": -1024,
                    "type": "integer"
                }               {

                    "number": 1,
                    "identifier": "ParameTypeReal",
                    "description": "",
                    "value": 0,
                    "minimum": 0,
                    "maximum": 4096,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 1,
                    "isOnline": true,
                    "formula": "10*\n10/",
                    "default": 100,
                    "type": "real"
                },
                {
                    "number": 2,
                    "identifier": "ParameTypeString",
                    "description": "",
                    "value": "text",
                    "access": "readWrite",
                    "enumeration": "Yes\nNo\nOk",
                    "isOnline": true,
                    "default": "default",
                    "type": "string"
                },
                {
                    "number": 3,
                    "identifier": "ParameTypeEnumeration",
                    "description": "",
                    "value": 0,
                    "access": "readWrite",
                    "enumeration": "Line\nMic+48V\nMic",
                    "isOnline": true,
                    "default": 0,
                    "type": "enum"
                },
                {
                    "number": 4,
                    "identifier": "ParameTypeBoolean",
                    "description": "",
                    "value": true,
                    "access": "readWrite",
                    "isOnline": true,
                    "default": false,
                    "type": "boolean"
                }
```

How we could do with you code to generate a json file with all the devices ?

Best regards
Youssef

adam-nielsen, 2019-12-23:
Modbus doesn't support any sort of enumeration like this.  The only way to get the information you need is to go through the manual that came with the device, and enter in the details you find.  It will be different for every device, and many devices store data in a non-standard way (e.g. a mix of big and little endian in the same value).

The only way you can write a working protocol converter is to allow the user to specify all the details for each device they want to use, and you must be willing to implement new data types as your users come across them.

yaacov, 2019-12-23:
To add on @adam-nielsen 's answer, if:
a - If you have a known and finite number of device types (type = have exactly the same parameters).
b - if you know all the possible address, e.g. 1..247 on serial or some small IP subnet.
Then you can probe each of the known possible address and map it to a device type.

closed 2020-03-14 by yaacov