yaacov/node-modbus-serial #315: Function 43/14 (Device Identification)

yadidgarbo, 2020-01-10:
Hello,

According to the documentation, function 14/43 can be executed.
Unfortunately I could not find an example for this.
Is it possible to get the device information from a ModBus TCP using 0x43/0x14?

yaacov, 2020-01-10:
> According to the documentation, function 14/43 can be executed.

Can you attach a reference ?

> Is it possible to get the device information from a ModBus TCP using 0x43/0x14?

IMHO no,

yadidgarbo, 2020-01-11:
Reference:
https://www.npmjs.com/package/modbus-serial

> These classes are implemented:
FC1 "Read Coil Status"
FC2 "Read Input Status"
FC3 "Read Holding Registers"
FC4 "Read Input Registers"
FC5 "Force Single Coil"
FC6 "Preset Single Register"
FC15 "Force Multiple Coil"
FC16 "Preset Multiple Registers"
FC43/14 "Read Device Identification" (server only)

yaacov, 2020-01-11:
> FC43/14 "Read Device Identification" **(server only)**

Thanks, I see, you can **serve** F43 and F14 using the TCP-server module [1}.
You can not read F43 or F14, you care welcome to add this functionality.

[1] https://github.com/yaacov/node-modbus-serial#modbustcp-server

yadidgarbo, 2020-01-11:
Thanks for the confirmation. Unfortunately I had expected this answer already.
Do you know of a node module which is able to read function 14/43.

Would help me a lot.

yaacov, 2020-01-11:
no :-(  

p.s.
You are welcome to implement one, and send a pull request ...

eeerin, 2020-03-12:
@yaacov you can close this issue now.

yaacov, 2020-03-12:
Fixed by #319

closing :birthday: :tada:

closed 2020-03-12 by yaacov

oicnal, 2020-03-14:
readDeviceIdentification fails (ETIMEDOUT) with RTUBuffered, but does work with RTU. I'm reading a array of Tracer solar charge controllers connected via RS485 to USB.