yaacov/node-modbus-serial #326: CRC error in serial modbus?

PAIDI-RAMESH, 2020-03-05:
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port
client.connectRTUBuffered("COM1", { baudRate: 9600,parity: 'none', dataBits : 8,stopBits : 1 }, read);

 function read() {
		client.readDiscreteInputs(1, 1)
		.then(function(data) {

		   console.log(data);
		   client.close();
		}).catch(err=>{console.log("error  "+err)});

}

yaacov, 2020-03-05:
Thank  you for the issue.

a. what error do you get ?
b. can you run with debug on ?
c. how often do you get the error ?

PAIDI-RAMESH, 2020-03-05:
[image]

yaacov, 2020-03-05:
can you send a pull request fixing it ?

exitrip, 2020-07-14:
I'm running this on a oscilloscope and can see that the modbus-serial code is transmitting 16 bit numbers for every register address and data field over the wire.  I can't tell what the module expects with the oscilloscope.

This would obviously mess up the CRCs if the device was using 8 bit registers.

(Un?)fortunately, I am definitely not a javascript programmer and have no business fixing this for others.  An FYI, the module definitely doesn't strictly enforce uint8_t in all the things.

alexszilagyi, 2020-07-17:
@exitrip : I'm a software developer and I'm facing the same thing on my project. Do you think you can help me with the debugging sessions? I'm not that good at electronics but did some studies previously. I'm only interested how did you've figure it out and to to debug actually.
I am looking forward to fix this and add a PR to the repo, @yaacov .

exitrip, 2020-07-17:
@alexszilagyi I ended up implementing everything using https://github.com/serialport/node-serialport directly and grabbing CRC javascript off a few websites.